patjak/facetimehd #217: Error on make

a8underscore, 2020-05-22:
Im on opensuse tumbleweed

here is the commands i run and error
`localhost:/home/nikita/Downloads/face/bcwc_pcie # make
make -C /lib/modules/5.6.12-1-default/build M=/home/nikita/Downloads/face/bcwc_pcie modules
make[1]: *** /lib/modules/5.6.12-1-default/build: No such file or directory.  Stop.
make: *** [Makefile:12: all] Error 2
localhost:/home/nikita/Downloads/face/bcwc_pcie # mkdir /lib/modules/5.6.12-1-default/build
localhost:/home/nikita/Downloads/face/bcwc_pcie # make
make -C /lib/modules/5.6.12-1-default/build M=/home/nikita/Downloads/face/bcwc_pcie modules
make[1]: Entering directory '/lib/modules/5.6.12-1-default/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/5.6.12-1-default/build'
make: *** [Makefile:12: all] Error 2
`

a8underscore, 2020-05-22:
im on macbook air 2015 btw

erratic-c-programmer, 2020-06-10:
Try https://github.com/patjak/bcwc_pcie/issues/221#issuecomment-641743001?

patjak, 2020-06-18:
Looks like your /lib/modules/KERNEL_VERSION/build directory isn't linking to the linux sources.
Did you forget to install the kernel-devel package?

slinkouswasalreadytaken, 2020-06-26:
```
[slinkous@MacBookAir bcwc_pcie]$ make
make -C /lib/modules/5.7.0-3-MANJARO/build M=/home/slinkous/bcwc_pcie modules
make[1]: Entering directory '/usr/lib/modules/5.7.0-3-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.7.0-3-MANJARO/build'
make: *** [Makefile:12: all] Error 2
[slinkous@MacBookAir bcwc_pcie]$ 
```
Same problem here. Using Manjaro on a MacBook Air 2017.
@erratic-c-programmer I tried doing that, but for some reason both sudo and su are giving me "you must be root to run this script."
@patjak I'm not sure how to do that. Using Manjaro, and pacman gives no results for that.

a8underscore, 2020-06-26:
Yea I don’t use macOS anymore I switched to opensuse tumbleweed and have no issue anymore, it seems to be macOS issue

erratic-c-programmer, 2020-06-27:
...this driver is for linux systems...
Please close this issue?

a8underscore, 2020-06-27:
Oh I’m so sorry I replied to the wrong GitHub issue email. I will try to compile this driver tommorow, sorry for any confusion

slinkouswasalreadytaken, 2020-06-27:
Still not working for me on Manjaro...?
I'm not using MacOS (I mean, I am right now, but when I got the make error)

erratic-c-programmer, 2020-06-30:
On arch(and manjaro) you have to install linux headers
```sudo pacman -S linux-headers```

slinkouswasalreadytaken, 2020-06-30:
Same error after reboot, looks like nothing was changed (0.00 to install).