patjak/facetimehd #221: Documentation

davehodg, 2020-06-03:
The documentation doesn't tell me how to install the driver.

And the link to a page describing how to use it is broken: http://chrisrjones.com/articles/How-to-get-the-broadcom-pci-1570-web-cam-working-in-Linux-using-qemu-with-OS-X-VM

erratic-c-programmer, 2020-06-10:
First `mkdir BUILD; cd BUILD`. Then copy the below code and paste it in a file `build.sh`. Then run it: `sudo bash build.sh`

```bash
#! /bin/bash
# Some inspiration from the AUR pkgbuild

if [ ! whoami = root ]; then
  echo 'You must be root to run this script.'
  exit 13
fi

pkgname=bcwc-pcie
pkgver=r245.82626d4
url="https://github.com/patjak/bcwc_pcie"
fwurl="https://github.com/patjak/facetimehd-firmware"
fwname=facetimehd-firmware
srcdir=src

rm -rf ${srcdir}

mkdir $srcdir

if [ ! -f /usr/lib/firmware/facetimehd/firmware.bin ]; then
  echo 'WARNING: Requires facetimehd-firmware, but it was not found. Installing it for you...'
  (cd src; git clone $fwurl $fwname)
  pushd ${srcdir}/${fwname} > /dev/null
  make
  mkdir /usr/lib/firmware/facetimehd/
  install -Dm 644 firmware.bin /usr/lib/firmware/facetimehd/firmware.bin
  popd > /dev/null
fi

(cd $srcdir; git clone $url $pkgname)

pushd $srcdir/$pkgname/ > /dev/null
for FILE in dkms.conf Makefile *.[ch]; do
  install -Dm 644 "$FILE" "/usr/src/${pkgname}-${pkgver}/$FILE"
done
popd > /dev/null

echo 'Creating modprobe config files'
echo 'facetimehd' > /etc/modules-load.d/bcwc-pcie.conf
echo 'blacklist bdc_pci' > /etc/modprobe.d/bcwc-pcie.conf

echo 'DKMS install for current kernel'
dkms add $pkgname -v $pkgver; dkms install $pkgname -v $pkgver
```

davehodg, 2020-06-17:
Two very minor comments:

The test for root failed despite me being root
dkms wasn't installed

Once I fixed both those things it worked! Nice one!

patjak, 2020-06-18:
The documentation for installing is available at https://github.com/patjak/bcwc_pcie/wiki/Get-Started
Perhaps this is misleading and should be called "Installation" instead of "Get started"?

slinkouswasalreadytaken, 2020-10-26:
Quick fix to erratic-c-programmer's program, replaced "=" with "!". 

```
#! /bin/bash
# Some inspiration from the AUR pkgbuild

if [ ! whoami != root ]; then
  echo 'You must be root to run this script.'
  exit 13
fi

pkgname=bcwc-pcie
pkgver=r245.82626d4
url="https://github.com/patjak/bcwc_pcie"
fwurl="https://github.com/patjak/facetimehd-firmware"
fwname=facetimehd-firmware
srcdir=src

rm -rf ${srcdir}

mkdir $srcdir

if [ ! -f /usr/lib/firmware/facetimehd/firmware.bin ]; then
  echo 'WARNING: Requires facetimehd-firmware, but it was not found. Installing it for you...'
  (cd src; git clone $fwurl $fwname)
  pushd ${srcdir}/${fwname} > /dev/null
  make
  mkdir /usr/lib/firmware/facetimehd/
  install -Dm 644 firmware.bin /usr/lib/firmware/facetimehd/firmware.bin
  popd > /dev/null
fi

(cd $srcdir; git clone $url $pkgname)

pushd $srcdir/$pkgname/ > /dev/null
for FILE in dkms.conf Makefile *.[ch]; do
  install -Dm 644 "$FILE" "/usr/src/${pkgname}-${pkgver}/$FILE"
done
popd > /dev/null

echo 'Creating modprobe config files'
echo 'facetimehd' > /etc/modules-load.d/bcwc-pcie.conf
echo 'blacklist bdc_pci' > /etc/modprobe.d/bcwc-pcie.conf

echo 'DKMS install for current kernel'
dkms add $pkgname -v $pkgver; dkms install $pkgname -v $pkgver
```

mabushey, 2020-12-12:
Ugg, the dead link was reported over half a year ago and there's no one left on this project to remove it. It sucks that this is a dead project, almost as much as how bad Apple products suck.

patjak, 2020-12-13:
@mabushey I don't understand your complaint. Where is the dead link located and do you really need that article specifically? I don't remember what it was about but the title says something about extracting the firmware from osx running inside qemu. That is not something you need to just install the driver and firmware. Did you look at the wiki? Please be more specific.

patjak, 2020-12-16:
@mabushey I just took a look at the Wiki and I now understand what you're talking about. The wiki is a mess. To clarify, everyone have access to modify the wiki so I have completely left it to the users to maintain. This is obviously not working.