patjak/facetimehd #224: macbook air 6,2 mid-2003 - facetimehd-firmware make error

monway, 2020-06-12:
Linux hell 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux
I tried this also on LMDE as well. excuse my ignorance, I'm not sure why this is happening.

`$ make
Checking dependencies for driver download...
/usr/bin/curl
/usr/bin/xzcat
/usr/bin/cpio

Downloading the driver, please wait...
mv: cannot stat 'System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface': No such file or directory
make: *** [Makefile:41: AppleCameraInterface] Error 1 
`

monway, 2020-06-12:
I seem to just manually extracted the firmware and ran the script again and it correctly installed. Odd the script doesn't itself download the files needed to extract. This was done manually. now after compiling the driver and installing it, cheese shows black screen. Frustrating.

koleesch, 2020-06-16:
I have the same issue... on an debian 10.4 (4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux)

patjak, 2020-06-18:
Works fine here. Perhaps there are some issues with the Apple download servers or your internet connection?

monway, 2020-10-20:
> This is probably just an issue with the certificates. You can update the certificates, or just add -k to the curl line in the Makefile, to fall back to insecure connections.

I found that you need innoextract, cab extract and curl. Addition to. To make this work.

maurolacy, 2020-10-21:
> I found that you need innoextract, cab extract and curl. Addition to. To make this work.

Well, of course you need `curl`. I suggest you to take a look at my PR above. It just required changing one char in the `Makefile` to make it work.

trnghiale, 2024-08-22:
I tried to adjust the Makefile but the error still appear. Can someone give some advice

trnghiale, 2024-08-22:
> > I found that you need innoextract, cab extract and curl. Addition to. To make this work.
> 
> Well, of course you need `curl`. I suggest you to take a look at my PR above. It just required changing one char in the `Makefile` to make it work.

I tried to adjust the Makefile but the error still appear. Can you give some advice?
Thank you.

trnghiale, 2024-08-23:
> > Can you give some advice?
> 
> Try downloading the file manually, to check that it's there in the Apple servers in the first place.
> 
> Then maybe adjust versions / URIs until it's valid.

Surprising, I ignore the warning and just keep following the steps, My webcam is recognized by guvcam but not cheese :)
Will check if it work on other application later

maurolacy, 2024-09-15:
> My webcam is recognized by guvcam but not cheese

Likely a permissions / group membership issue of you Linux distro.