patjak/facetimehd #227: Failure after installation (error -2)

quickTwix898, 2020-07-31:
Hello, I am running an up-to-date installation Fedora Linux 32 on a Macbook Pro ([Retina, 15-inch, Mid 2015, no touchbar](https://support.apple.com/kb/sp719?locale=en_US)). I followed all instructions (firmware extraction, getting started on fedora). I read all warnings for the build and install and it seems that nothing out of the ordinary has happened. 
#### Problem
However, when I open cheese or any other application that requires webcam, no device is detected. 
#### Expected 
Since I followed the instructions the webcam should be able to work. 
#### Evidence of Problem 
```
[user@computer /]$ dmesg | grep facetimehd
[    1.818613] facetimehd: loading out-of-tree module taints kernel.
[    1.818650] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[    1.819416] facetimehd 0000:05:00.0: Found FaceTime HD camera with device id: 1570
[    1.819586] facetimehd 0000:05:00.0: Setting 64bit DMA mask
[    1.824681] facetimehd 0000:05:00.0: S2 PCIe link init succeeded
[    1.824708] facetimehd 0000:05:00.0: Refclk: 25MHz (0xa)
[    1.834719] facetimehd 0000:05:00.0: PLL reset finished
[    1.834720] facetimehd 0000:05:00.0: Waiting for S2 PLL to lock at 450 MHz
[    1.834732] facetimehd 0000:05:00.0: S2 PLL is locked after 10 us
[    1.844738] facetimehd 0000:05:00.0: S2 PLL is in bypass mode
[    1.864773] facetimehd 0000:05:00.0: DDR40 PHY PLL locked on safe settings
[    1.864787] facetimehd 0000:05:00.0: STRAP valid
[    1.864788] facetimehd 0000:05:00.0: Configuring DDR PLLs for 450 MHz
[    1.864795] facetimehd 0000:05:00.0: DDR40 PLL is locked after 0 us
[    1.864806] facetimehd 0000:05:00.0: First DDR40 VDL calibration completed after 2 us
[    1.864817] facetimehd 0000:05:00.0: Second DDR40 VDL calibration completed after 2 us
[    1.864818] facetimehd 0000:05:00.0: Using step size 150
[    1.864823] facetimehd 0000:05:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[    1.864828] facetimehd 0000:05:00.0: Virtual VTT enabled
[    1.885348] facetimehd 0000:05:00.0: S2 DRAM memory address: 0x22159559
[    1.885361] facetimehd 0000:05:00.0: Rewrite DDR mode registers succeeded
[    1.885561] facetimehd 0000:05:00.0: Full memory verification succeeded! (0)
[    1.885741] facetimehd 0000:05:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[    1.885805] facetimehd: probe of 0000:05:00.0 failed with error -2
```

#### Additional Information
```
[user@computer /]$ uname -a
Linux localhost.localdomain 5.7.10-201.fc32.x86_64 #1 SMP Thu Jul 23 00:58:39 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```
[user@computer /]$ lsmod | grep facetime
facetimehd            102400  0
videobuf2_dma_sg       16384  1 facetimehd
videobuf2_v4l2         28672  1 facetimehd
videobuf2_common       57344  2 videobuf2_v4l2,facetimehd
videodev              270336  3 videobuf2_v4l2,facetimehd,videobuf2_common
```

Here is the list of the exact commands I followed, in strict chronological order. It should be identical to that described in the wiki: https://pastebin.com/bn88mfJK

frgt10, 2020-07-31:
Try adding a config for Dracut:

```bash
echo 'install_items+=" /usr/lib/firmware/facetimehd/firmware.bin "' >> /etc/dracut.conf.d/facetimehd.conf
dracut -f
```
And reboot.

quickTwix898, 2020-07-31:
Thank you, this fixed the problem. I will close the issue.

closed 2020-07-31 by quickTwix898

Bios-Marcel, 2021-11-23:
Hey, I was trying the fix, but got this:

```
dracut-install: ERROR: installing '/usr/lib/firmware/facetimehd/firmware.bin'
```