patjak/facetimehd #229: Full memory verification failed on MBA7,2

Rijul-Ahuja, 2020-10-01:
Hello

I am using a Macbook Air recognized as 7,2 on Pop-OS 20.04 (Kernel 5.4.0).
```
$ sudo dmidecode -s system-product-name
MacBookAir7,2
$ uname -a
Linux pop-os 5.4.0-7642-generic #46~1598628707~20.04~040157c-Ubuntu SMP Fri Aug 28 18:02:16 UTC  x86_64 x86_64 x86_64 GNU/Linux
```

The steps that I followed are:
```
$ sudo apt install xz-utils curl cpio
$ sudo modprobe -r bdc_pci
$ sudo modprobe -a v4l2-common videobuf2-dma-sg videobuf2-core videobuf2-v4l2
$ cd ~/Desktop
$ mkdir camera
$ cd camera
$ git clone https://github.com/patjak/bcwc_pcie.git
$ cd bcwc_pcie
$ git clone https://github.com/patjak/facetimehd-firmware.git
$ cd facetimehd-firmware
$ make
$ sudo make install
$ cd ..
$ make
$ sudo checkinstall
$ sudo depmod
$ sudo modprobe facetimehd
$ mplayer tv://
```

The last line throws the following error:
```
MPlayer 1.3.0 (Debian), built with gcc-9 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: unable to open '/dev/video0': No such file or directory
v4l2: ioctl set mute failed: Bad file descriptor
v4l2: 0 frames successfully processed, 0 frames dropped.

Exiting... (End of file)
```

This indicates that `/dev/video{,0,1}` was not created, which is true
```
$ ls -la /dev/vid*
ls: cannot access '/dev/vid*': No such file or directory
```

The `$ modprobe -a ...` line also spits out something concerning:
```
modprobe: WARNING: Module v4l2-common not found in directory /lib/modules/5.4.0-7642-generic
modprobe: WARNING: Module videobuf2-core not found in directory /lib/modules/5.4.0-7642-generic
```

Looking at `dmesg`, I see:
```
[ 2220.456678] mc: Linux media interface: v0.10
[ 2220.464523] videodev: Linux video capture interface: v2.00
[ 2235.551139] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 2235.551236] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 2235.556698] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 2235.556727] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 2235.566660] facetimehd 0000:02:00.0: PLL reset finished
[ 2235.566661] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 2235.566673] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[ 2235.576599] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[ 2235.596479] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 2235.596493] facetimehd 0000:02:00.0: STRAP valid
[ 2235.596494] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 2235.596513] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 2235.596523] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 2235.596535] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 2235.596536] facetimehd 0000:02:00.0: Using step size 156
[ 2235.596542] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[ 2235.596548] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 2235.616909] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 2235.616923] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 2235.617117] facetimehd 0000:02:00.0: Full memory verification failed! (65535)
[ 2235.731936] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[ 2245.826276] facetimehd 0000:02:00.0: Init failed! No wake signal
[ 2245.826412] facetimehd: probe of 0000:02:00.0 failed with error -5
```

To further trace this, I try 
```
$ sudo find / -name facetimehd.ko
/usr/lib/modules/5.4.0-7642-generic/extra/facetimehd.ko
$ sudo modprobe -r facetimehd
$ sudo insmod /usr/lib/modules/5.4.0-7642-generic/extra/facetimehd.ko dyndbg=pmf
insmod: ERROR: could not insert module /usr/lib/modules/5.4.0-7642-generic/extra/facetimehd.ko: Unknown symbol in module
$ dmesg
[ 2739.899659] facetimehd: Unknown symbol vb2_queue_init (err -2)
[ 2739.899682] facetimehd: Unknown symbol vb2_dma_sg_memops (err -2)
[ 2739.899706] facetimehd: Unknown symbol video_ioctl2 (err -2)
[ 2739.899729] facetimehd: Unknown symbol vb2_ioctl_streamoff (err -2)
[ 2739.899749] facetimehd: Unknown symbol vb2_ioctl_expbuf (err -2)
[ 2739.899772] facetimehd: Unknown symbol vb2_ops_wait_finish (err -2)
[ 2739.899791] facetimehd: Unknown symbol v4l2_ctrl_handler_init_class (err -2)
[ 2739.899814] facetimehd: Unknown symbol vb2_ioctl_querybuf (err -2)
[ 2739.899835] facetimehd: Unknown symbol v4l2_fh_open (err -2)
[ 2739.899863] facetimehd: Unknown symbol video_devdata (err -2)
[ 2739.899881] facetimehd: Unknown symbol vb2_fop_release (err -2)
[ 2739.899902] facetimehd: Unknown symbol vb2_plane_cookie (err -2)
[ 2739.899922] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err -2)
[ 2739.899943] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err -2)
[ 2739.899964] facetimehd: Unknown symbol vb2_buffer_done (err -2)
[ 2739.899983] facetimehd: Unknown symbol v4l2_ctrl_subscribe_event (err -2)
[ 2739.900002] facetimehd: Unknown symbol video_unregister_device (err -2)
[ 2739.900022] facetimehd: Unknown symbol vb2_ioctl_qbuf (err -2)
[ 2739.900041] facetimehd: Unknown symbol vb2_fop_mmap (err -2)
[ 2739.900060] facetimehd: Unknown symbol video_device_alloc (err -2)
[ 2739.900081] facetimehd: Unknown symbol vb2_fop_read (err -2)
[ 2739.900099] facetimehd: Unknown symbol v4l2_device_register (err -2)
[ 2739.900130] facetimehd: Unknown symbol __video_register_device (err -2)
[ 2739.900149] facetimehd: Unknown symbol vb2_ops_wait_prepare (err -2)
[ 2739.900172] facetimehd: Unknown symbol vb2_ioctl_streamon (err -2)
[ 2739.900190] facetimehd: Unknown symbol vb2_fop_poll (err -2)
[ 2739.900209] facetimehd: Unknown symbol v4l2_ctrl_new_std (err -2)
[ 2739.900227] facetimehd: Unknown symbol v4l2_ctrl_handler_free (err -2)
[ 2739.900245] facetimehd: Unknown symbol v4l2_device_unregister (err -2)
[ 2739.900264] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err -2)
[ 2739.900284] facetimehd: Unknown symbol video_device_release (err -2)
[ 2739.900305] facetimehd: Unknown symbol v4l2_event_unsubscribe (err -2)
[ 2739.900325] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err -2)
```

This appears to be a combination of two errors: Full memory verification failed and unknown symbol v4l2* when loading the driver. Let me know if there is some other information I can help out with, thanks!

Rijul-Ahuja, 2020-10-02:
The issue is resolved by doing this step https://github.com/patjak/bcwc_pcie/wiki/Extracting-the-sensor-calibration-files and reloading the driver.
```
sudo modprobe -r facetimehd
sudo modprobe facetimehd
```

Rijul-Ahuja, 2020-10-07:
This issue has now reappeared even though the dat files are in place.

patjak, 2020-12-16:
It seem the memory verification fails on your webcam. I never finished implementing the memory controller calibration since it's quite complex and normally works in it's default state. Sorry, you might be out of luck here.