patjak/facetimehd #230: Problem with new kernel 5.4.0-51

visavi, 2020-10-15:
Ubuntu 20.04
`
$ make
make -C /lib/modules/5.4.0-51-generic/build M=/home/bcwc_pcie modules
make[1]: вход в каталог «/usr/src/linux-headers-5.4.0-51-generic»
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: выход из каталога «/usr/src/linux-headers-5.4.0-51-generic»

$ sudo make install
make -C /lib/modules/5.4.0-51-generic/build M=/home/bcwc_pcie modules_install
make[1]: вход в каталог «/usr/src/linux-headers-5.4.0-51-generic»
  INSTALL /home/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-51-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: выход из каталога «/usr/src/linux-headers-5.4.0-51-generic»
`
there was a problem with ssl, I ignored it and everything worked fine
But after the update there was a new problem  missing 'System.map' file
and the camera no longer works

Before that I have been updated many times already, everything was always successful

kclarkr, 2020-10-18:
I have the same problem.  I created a symlink in /boot to the matching System.map-5.4.0-51-generic file and it still didn't work.

`lrwxrwxrwx 1 root root       27 Oct 18 11:30 System.map -> System.map-5.4.0-51-generic
-rw------- 1 root root  4743112 Sep 10 05:12 System.map-5.4.0-48-generic
-rw------- 1 root root  4743490 Oct  5 08:34 System.map-5.4.0-51-generic
`

kclarkr, 2020-10-20:
I upgraded to kernel & src 5.4.0.52 and had the same issue with the System.map error.  I created a symlink for System.map in /usr/src/linux-headers-5.4.0.52-generic to pointing to /boot/System.map-5.4.0.52-generic and the sudo make install command worked.

visavi, 2020-10-22:
updated to 5.4.0-52 the error is the same, but the camera is already working

toger5, 2020-12-22:
Are there already any solutions to that error?
For me it looks like this:
```
 sudo make
make -C /lib/modules/5.8.15-301.fc33.x86_64/build M=/home/$USER/Projects/bcwc_pcie modules
make[1]: *** /lib/modules/5.8.15-301.fc33.x86_64/build: No such file or directory.  Stop.
make: *** [Makefile:12: all] Error 2
```

cgalo5758, 2022-03-20:
> I upgraded to kernel & src 5.4.0.52 and had the same issue with the System.map error. I created a symlink for System.map in /usr/src/linux-headers-5.4.0.52-generic to pointing to /boot/System.map-5.4.0.52-generic and the sudo make install command worked.
@kclarkr, I want to try doing this. When you say you created a symlink for `System.map` in `/usr/src/linux-headers-5.4.0.52-generic` pointing to `/boot/System.map-5.4.0.52`. How did you do it?

Where in `/usr/src/linux-headers-...` did you create the symlink and what did you name it?

---
This is what I got when I tried `sudo make install`
```
~/bcwc_pcie$ sudo make install 
make -C /lib/modules/5.10.0-12-amd64/build M=/home/christian/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-12-amd64'
  INSTALL /home/christian/bcwc_pcie/facetimehd.ko
  DEPMOD  5.10.0-12-amd64
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-12-amd64'
```

--- 
I added a symlink to `/usr/src/linux-headers-...`but I end up getting tons of warnings like:
```
...
depmod: WARNING: /lib/modules/5.10.0-12-amd64/extra/facetimehd.ko needs unknown symbol release_resource
depmod: WARNING: /lib/modules/5.10.0-12-amd64/extra/facetimehd.ko needs unknown symbol debugfs_create_dir
depmod: WARNING: /lib/modules/5.10.0-12-amd64/extra/facetimehd.ko needs unknown symbol pci_release_region
depmod: WARNING: /lib/modules/5.10.0-12-amd64/extra/facetimehd.ko needs unknown symbol __kmalloc
...

```

Not sure what to do.

kash-raman, 2022-04-23:
Im having similar problem with System map, @patjak  any thoughts. I'm on Pop OS 20.04 LTS/MacBookPro late 2013 
```

facetimehd-firmware/bcwc_pcie# make install
make -C /lib/modules/5.16.19-76051619-generic/build M=/tech/apps/facetimehd-firmware/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.16.19-76051619-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.16.19-76051619-generic/extra/facetimehd.ko
  SIGN    /lib/modules/5.16.19-76051619-generic/extra/facetimehd.ko
  DEPMOD  /lib/modules/5.16.19-76051619-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.16.19-76051619-generic'
```

james2doyle, 2024-05-12:
This worked for me to get past this error:

```sh
cd /usr/src/linux-headers-6.8.0-31-generic
sudo ln -s /boot/System.map-6.8.0-31-generic $PWD/System.map
```

ca4ti, 2024-11-23:
> This worked for me to get past this error:
> 
> ```shell
> cd /usr/src/linux-headers-6.8.0-31-generic
> sudo ln -s /boot/System.map-6.8.0-31-generic $PWD/System.map
> ```
> 
> You should now be able to run `make install` without the error

lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.10
Release:	24.10
Codename:	oracular

ln -sf /boot/System.map-$(uname -r) /usr/src/linux-headers-$(uname -r)/System.map