patjak/facetimehd #231: Module is loaded but not working on 5.4.70 (but working on 5.4.3)

jze1, 2020-10-22:
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.533010] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.533428] facetimehd 0000:02:00.0: Setting 64bit DMA mask
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.540421] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.540449] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.550394] facetimehd 0000:02:00.0: PLL reset finished
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.550396] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.550407] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.560362] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580279] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580293] facetimehd 0000:02:00.0: STRAP valid
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580294] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580301] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580309] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580319] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580320] facetimehd 0000:02:00.0: Using step size 152
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580325] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.580329] facetimehd 0000:02:00.0: Virtual VTT enabled
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.600725] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.600738] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.600932] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
Oct 22 20:37:58 jz-MacBookPro kernel: [ 1173.715868] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.874017] facetimehd 0000:02:00.0: ISP woke up after 0ms
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.874022] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.874046] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.884025] facetimehd 0000:02:00.0: ISP second int after 0ms
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.884027] facetimehd 0000:02:00.0: Channel description table at 00800000
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894209] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894210] facetimehd 0000:02:00.0: Enabling interrupts
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894457] FWMSG: 
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894489] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894498] FWMSG: 
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894512] FWMSG:   Restart count : 0
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894539] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894557] FWMSG:   System Clock : 200000000 Hz
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894574] FWMSG:   Processor mode : SUPERVISOR
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894594] FWMSG:   Cache architecture type : SEPARATE
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894610] FWMSG:   Cache type : WRITEBACK
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894650] FWMSG:   Data Cache Line : 32 [0x20]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894667] FWMSG:   Boot arguments entries : 2
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894764] FWMSG:     0000: 0x00000000 0x00000000
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894789] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894808] FWMSG:   Main memory :   base : 0x00000000
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894833] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894858] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894884] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.894907] FWMSG:   Shared window : base : 0x00800000 [static wiring]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895034] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895052] FWMSG:   Shared memory : base : 0x00800000
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895079] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895108] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895268] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895289] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895309] FWMSG:   Heap allocated space : 94656 [0x000171c0]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895336] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895353] FWMSG:   Application : adc [release]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895373] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895390] FWMSG:   Release : S2ISP-01.43.00
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895405] FWMSG:   H4ISPAPPLE : 11536
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895419] FWMSG:   H4ISPCD : 4081
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895432] FWMSG:   ffw : 4143
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895455] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895465] FWMSG: 
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895496] FWMSG: -------------------------------------------------------------------------------
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.895505] FWMSG: 
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.964732] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.964974] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965247] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965408] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965440] ------------[ cut here ]------------
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965449] WARNING: CPU: 2 PID: 4519 at drivers/media/v4l2-core/v4l2-dev.c:863 __video_register_device+0x39e/0x600 [videodev]
[... 29 lines cut ...]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965527]  ? 0xffffffffc0cdf000
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965528]  __pci_register_driver+0x57/0x60
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965531]  fthd_pci_driver_init+0x23/0x1000 [facetimehd]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965534]  do_one_initcall+0x4a/0x1fa
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965536]  ? _cond_resched+0x19/0x30
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965538]  ? kmem_cache_alloc_trace+0x16c/0x240
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965541]  do_init_module+0x62/0x250
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965542]  load_module+0x10b8/0x1200
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965545]  __do_sys_finit_module+0xbe/0x120
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965546]  ? __do_sys_finit_module+0xbe/0x120
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965549]  __x64_sys_finit_module+0x1a/0x20
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965551]  do_syscall_64+0x57/0x190
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965552]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[... cut ...]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965558] R13: 000055ff0bd0ded0 R14: 0000000000040000 R15: 000055ff0bd0dd60
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965560] ---[ end trace 2d5fc039f0775701 ]---
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.965716] FWMSG: [ISP] CMD = 0x0001 [CISP_CMD_STOP]
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.975760] facetimehd 0000:02:00.0: fthd_hw_deinit
Oct 22 20:37:59 jz-MacBookPro kernel: [ 1173.975965] facetimehd: probe of 0000:02:00.0 failed with error -22

serokellcao, 2020-12-11:
Can confirm that it doesn't work on 5.8.0 either.

Diadochokinetic, 2020-12-15:
Doesn't work on 5.4.0-58-generic either.

patjak, 2020-12-15:
Do you all get the same warning?

alexshpilkin, 2020-12-15:
I’m on MBP 11,1 using kernel 5.9.14.arch1-1 and the firmware appears to be loading successfully using the latest git (and the camera subsequently works across qv4l2, Cheese, Chromium, and Skype for Linux); the messages are almost identical to the ones in the bug report up to `Direct firmware load failed ...` (I have `FWMSG: [ISP] CH = 0 CMD = 0x0111 [CISP_CMD_CH_SET_FILE_LOAD]` instead, and `Using step size 156` instead of `Using step size 152` before that). So it _might_ potentially be a distro thing, given the wide range of kernel versions reported here. (I can try building and booting an older kernel if necessary.)

patjak, 2020-12-16:
I just tried it on opensuse tumbleweed with 5.9 kernel. Everything works as expected. Any other information you can provide that might help pinpoint the problem?

@alexshpilkin yes please try 5.8.0 as reported by @serokellcao

jze1, 2020-12-16:
I tried 5.9.12 mainline kernel for ubuntu, but since I am using it normally connected to thunderbolt monitor with keyboard and mouse attached to the monitor, it is not usable for me as it seems like the keyboard and mouse data are transferred with a telegraph speed (big cursor lag and keyboard misses some keystrokes).

But with 5.4.70 it at least recognises the facetime camera in the thunderbolt monitor.

patjak, 2020-12-16:
@jze1 sounds like this is a problem with Thunderbolt and not the facetimehd driver. What does ```lspci -vvv -d 14e4:1570``` say about the device with the non-working kernel?

jze1, 2020-12-17:
02:00.0 Multimedia controller: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera
	Subsystem: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at c1700000 (64-bit, non-prefetchable) [size=64K]
	Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
	Region 4: Memory at c1600000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=2 PME-
	Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [68] Vendor Specific Information: Len=44 <?>
	Capabilities: [ac] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 10.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <32us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE#
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [13c v1] Device Serial Number 00-00-00-ff-ff-00-00-00
	Capabilities: [150 v1] Power Budgeting <?>
	Capabilities: [160 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [1b0 v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Capabilities: [220 v1] #15
	Kernel modules: bdc_pci

patjak, 2020-12-17:
I can see you have the bdc_pci module loaded for the device. This is incorrect.
Run ```modprobe -r bdc_pci```and then ```modprobe facetimehd``` to reload the facetimehd driver

Diadochokinetic, 2020-12-26:
Hi,

I tried to unload bdc_pci and load facetimehd and ran into the following problems:

After running `modprobe -r bdc_pci`, I checked with `cat /proc/modules | grep bdc` and the module was not listed. But when running `lspci -vvv -d 14e4:1570` the kernel module was still bdc_pci. After rebooting the machine facetimehd was not loaded and bdc_pci was loaded. Is there anything else I should try?

Diadochokinetic, 2020-12-27:
Hello again,

I think I have located the problem. After updating the "blacklist bdc_pci" entry from /etc/modprobe.d/dkms.conf was missing. This leads to bdc_pci being loaded at reboot, preventing the facetimehd module from working.

jze1, 2020-12-28:
lsmod does not show bdc_pci. So I guess it is built-in in this version of kernel. Will check how to disable it......

Diadochokinetic, 2021-01-05:
I've just updated from 5.4.0-58-generic to 5.4.0-59-generic and had no issues. The camera is working, bdc_pci is blacklisted and the facetimehd module is loaded.

patjak, 2021-01-27:
Great. I have a patch going into the stable kernels that finally completely disables bdc_pci so no blacklisting will be necessary when this hits peoples distros.

closed 2021-01-27 by patjak