patjak/facetimehd #233: firmware load failed with error -2, Ubuntu 20.10

jaxjexjox, 2020-12-16:
Hi there,

Trying to follow instructions (from 16.04, sorry) to get this working, with no dice on 20.10.
[ 1839.777201] facetimehd 0000:03:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[ 1839.777270] facetimehd: probe of 0000:03:00.0 failed with error -2

"sudo modprobe facetimehd" appeared to return nothing for me unfortunately.

patjak, 2020-12-16:
You're missing the firmware files. See https://github.com/patjak/bcwc_pcie/wiki/Get-Started#firmware-extraction

jaxjexjox, 2020-12-20:
Thank you for your help!  Sorry I must have missed it in the guide.
I'll say the picture quality is a tad grainy and the driver does appear to 'die' from time to time, but it's free, so thank you.

Hopefully in time the stability will improve.  Thanks.

patjak, 2020-12-21:
Yes, image quality on these cameras aren't great, but it shouldn't die. What does dmesg say when it "dies"? Which Mac model is this?

jaxjexjox, 2020-12-24:
Hi,

She's stopped behaving entirely now, I'm unsure if my work on getting the audio working is hindering in some way.
Macbook 13,1 A1708

Here is a pastebin of the entire dmesg.
https://pastebin.com/0fcnP8HJ
(Sorry best I can think of?)

aijanai, 2020-12-25:
Try this:
https://gist.github.com/aijanai/8c01167354b48996641c3544c267d394

I had to revamp my script to have it work on Ubuntu 20.04 with latest kernel, but now it should work.

patjak, 2021-01-27:
The upstream bdc_pci driver is going away. This will solve the conflict of bdc binding to the webcam. I have a patch going to the stable kernels that will also disable bdc. Hopefully this will arrive in peoples kernels soon.

closed 2021-01-27 by patjak