patjak/facetimehd #234: Camera is still not working - Gentoo

rubusarbaro, 2020-12-18:
Hi, I'm using Gentoo Linux on my MacBook Air 7,1
I installed the driver from the toaster overlay, but the camera is still not working.
I want to know if I have to recompile the kernel, because I got the following error while `modprobe facetimehd` or `modprobe -r bdc_pci`
[image]

patjak, 2020-12-21:
I'm not familiar with Gentoo or the toaster overlay but it seems the driver isn't installed for the running kernel (or perhaps not at all).

robinmarlow, 2021-03-17:
Toaster overlay is working for me.
You can see where the module is being installed by looking at:
`find /lib -name facetimehd.ko`
compare that to the name of your running kernel
`uname -a`
if you need to change to make them match it would be:
`eselect kernel list`
`eselect kernel set x`
Then you might have to re-install the module again:
`emerge -vp bcwc_pcie`