patjak/facetimehd #235: Having some trouble on a Macbook Air Mid 2011

Vitorvlv, 2020-12-21:
hey everyone, would appreciate some help on getting this working on a old macbook air.

I believe I have a compatible model with the driver, please correct me if I'm wrong.

I'm on Manjaro. I've installed both the driver and the firmware through the AUR, there were no errors. My issue is that I'm getting no video from the webcam, I've tried through the browser (google meets) and with some webcam software (kamoso and cheese), all I get is a black screen.

I'm attaching the output of lshw

With this being the relevant part, I believe:
`              *-usb:1
                   description: Video
                   product: FaceTime Camera (Built-in)
                   vendor: Apple Inc.
                   physical id: 2
                   bus info: usb@1:2
                   version: 6.46
                   serial: CCGEA900NADL8LLX
                   capabilities: usb-2.00
                   configuration: driver=uvcvideo maxpower=500mA speed=480Mbit/s`

[output lshw macbook air mid 2011.txt](https://github.com/patjak/bcwc_pcie/files/5726124/output.lshw.macbook.air.mid.2011.txt)

therealtear, 2020-12-21:
im also on manjaro receiving this error with a macbook pro 2015

Diadochokinetic, 2021-01-06:
Hi,

can you verify, that the module is loaded? What does `lsmod | grep facetimehd` output?

I'm on an old MacBook Air aswell and it's working fine. I did not install it from a repo but cloned the git repo and compiled from source.

therealtear, 2021-01-06:
> Hi,
> 
> can you verify, that the module is loaded? What does `lsmod | grep facetimehd` output?
> 
> I'm on an old MacBook Air aswell and it's working fine. I did not install it from a repo but cloned the git repo and compiled from source.

Installing from pamac and compiling myself both corrupted the kernel and completely fucked my manjaro install I just started reinstalling macos about 5 minutes ago

Vitorvlv, 2021-01-13:
> Hi,
> 
> can you verify, that the module is loaded? What does `lsmod | grep facetimehd` output?
> 
> I'm on an old MacBook Air aswell and it's working fine. I did not install it from a repo but cloned the git repo and compiled from source.

Hi, I apologize for the wait, here's the output:

```

facetimehd             98304  0
videobuf2_dma_sg       16384  1 facetimehd
videobuf2_v4l2         36864  2 facetimehd,uvcvideo
videobuf2_common       65536  3 videobuf2_v4l2,facetimehd,uvcvideo
videodev              290816  4 videobuf2_v4l2,facetimehd,uvcvideo,videobuf2_common
```

It looks like the module is loaded

Vitorvlv, 2021-01-22:
I installed MacOS again and it turns out it's not being recognized there either. I can only assume it's a hardware problem.

closed 2021-01-22 by Vitorvlv