patjak/facetimehd #236: Does not compile on Kernel 5.7.1-050701 [Ubuntu 16.04]

addiks, 2021-01-13:
Hello,

I have just upgraded from the official Kernel 4.15 that is shipped with Ubuntu 16.04.7 to a newer mainline kernel 5.7.1 due to better support for thunderbolt display hotplugging. On the old kernel (4.15), this driver worked perfectly after compiling and installing. With the new kernel (5.7.1) this driver refuses to compile.
This is the log when trying to execute make:
```

gerrit@gerrit-macbook:~/git/bcwc_pcie$ make
make -C /lib/modules/5.7.1-050701-generic/build M=/home/gerrit/git/bcwc_pcie modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.7.1-050701-generic“ wird betreten
  CC [M]  /home/gerrit/git/bcwc_pcie/fthd_ddr.o
In file included from <command-line>:0:0:
./arch/x86/include/asm/segment.h: In function ‘vdso_read_cpunode’:
././include/linux/compiler_types.h:201:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/alternative.h:240:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile (ALTERNATIVE(oldinstr, newinstr, feature) \
  ^
./arch/x86/include/asm/segment.h:266:2: note: in expansion of macro ‘alternative_io’
  alternative_io ("lsl %[seg],%[p]",
  ^
In file included from ./arch/x86/include/asm/barrier.h:5:0,
                 from ./include/linux/compiler.h:256,
                 from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:83,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/random.h:10,
                 from /home/gerrit/git/bcwc_pcie/fthd_ddr.c:20:
./arch/x86/include/asm/alternative.h:108:2: error: expected identifier or ‘(’ before string constant
  "# ALT: oldnstr\n"      \
  ^
./arch/x86/include/asm/alternative.h:159:2: note: in expansion of macro ‘OLDINSTR’
  OLDINSTR(oldinstr, 1)      \
  ^
./arch/x86/include/asm/alternative.h:240:23: note: in expansion of macro ‘ALTERNATIVE’
  asm_inline volatile (ALTERNATIVE(oldinstr, newinstr, feature) \
                       ^
./arch/x86/include/asm/segment.h:266:2: note: in expansion of macro ‘alternative_io’
  alternative_io ("lsl %[seg],%[p]",
  ^
In file included from <command-line>:0:0:
././include/linux/compiler_types.h:201:24: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/alternative.h:240:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile (ALTERNATIVE(oldinstr, newinstr, feature) \
  ^
./arch/x86/include/asm/segment.h:266:2: note: in expansion of macro ‘alternative_io’
  alternative_io ("lsl %[seg],%[p]",
  ^
./arch/x86/include/asm/page_64.h: In function ‘clear_page’:
././include/linux/compiler_types.h:201:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
./arch/x86/include/asm/alternative.h:256:2: note: in expansion of macro ‘asm_inline’
  asm_inline volatile (ALTERNATIVE_2("call %P[old]", "call %P[new1]", feature1,\
  ^
./arch/x86/include/asm/page_64.h:49:2: note: in expansion of macro ‘alternative_call_2’
  alternative_call_2(clear_page_orig,
  ^
In file included from ./arch/x86/include/asm/barrier.h:5:0,
                 from ./include/linux/compiler.h:256,
                 from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:83,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/random.h:10,
                 from /home/gerrit/git/bcwc_pcie/fthd_ddr.c:20:
./arch/x86/include/asm/alternative.h:128:2: error: expected identifier or ‘(’ before string constant
  "# ALT: oldinstr2\n"         \
  ^
[... 3140 lines cut ...]
./include/asm-generic/bug.h:115:3: note: in expansion of macro ‘__WARN’
   __WARN();      \
   ^
./include/media/v4l2-subdev.h:974:6: note: in expansion of macro ‘WARN_ON’
  if (WARN_ON(pad >= sd->entity.num_pads))
      ^
scripts/Makefile.build:266: die Regel für Ziel „/home/gerrit/git/bcwc_pcie/fthd_ddr.o“ scheiterte
make[2]: *** [/home/gerrit/git/bcwc_pcie/fthd_ddr.o] Fehler 1
Makefile:1729: die Regel für Ziel „/home/gerrit/git/bcwc_pcie“ scheiterte
make[1]: *** [/home/gerrit/git/bcwc_pcie] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-5.7.1-050701-generic“ wird verlassen
Makefile:12: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
gerrit@gerrit-macbook:~/git/bcwc_pcie$ 
```

At first glance, it looks like something with inline assembler instructions changed in the newer kernel:
```
...
././include/linux/compiler_types.h:201:24: error: expected ‘(’ before ‘__inline’
 #define asm_inline asm __inline
                        ^
...
```
Maybe it is something similar to [this](https://github.com/iovisor/bcc/issues/2546) issue? I will try to investigate further, but I am really not experienced with these things.

ukn, 2021-01-20:
Got the same problem after moving to kernel 5.4.0-62.
I found this https://www.mail-archive.com/ubuntu-bugs@lists.ubuntu.com/msg5671020.html

I temporarily commented out the line with `CONFIG_CC_HAS_ASM_INLINE` in `/usr/src/linux-headers-$(uname -r)/include/generated/autoconf.h` and complication worked.

addiks, 2021-01-20:
Thank you very much. This did work for me.

closed 2021-01-20 by addiks