patjak/facetimehd #238: Macbook 14,2 | ElementaryOS

starlightromero, 2021-02-01:
Upon `make install` in `bcwc_pcie` I receive the following error
```
make -C /lib/modules/5.4.0-65-generic/build M=/home/starlight/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-65-generic'
  INSTALL /home/starlight/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-65-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-65-generic'
```

alexshpilkin, 2021-02-01:
If you squint at it a little, this message tells you exactly what’s wrong; and in any case it has nothing to do with this module in particular and everything to do with building out-of-tree (i.e. not-bundled-with-Linux) modules in general.

Specifically, you seem to be using a machine with Secure Boot enabled, in which case both the kernel image and any module files need to be signed by a recognized key. (Signing of kernel images is enforced by the boot loader; signing of module files is then enforced by the kernel.) Normally your distribution takes care of this, but now that you’re building a module yourself, you’ll need to sign it yourself, too.

The instructions for setting up module signing on Ubuntu (of which elementary OS is a derivative) can be found [in the Ubuntu blog][1]. I don’t use Ubuntu myself so I won’t be able to help you in any more detail, but people in the usual Ubuntu support channels ([IRC][2], [mailing lists][3], [Q&A][4], [forums][5]) probably will.

Alternatively, if that is permitted by your hardware and acceptable to you security-wise (!), you can just disable Secure Boot altogether and forget about all this.

[1]: https://ubuntu.com/blog/how-to-sign-things-for-secure-boot
[2]: https://wiki.ubuntu.com/IRC/ChannelList
[3]: https://lists.ubuntu.com/
[4]: https://askubuntu.com/
[5]: https://ubuntuforums.org/

starlightromero, 2021-02-02:
I will look into the links you provided. Thank you!

closed 2021-02-02 by starlightromero