patjak/facetimehd #239: Macbook wont sleep/hibernate properly when lid closed

auyer, 2021-02-05:
Hi. This module works in respect of having webcam functionality, but it makes hibernation on lid-close hang at least half of the times.
What happens if that the monitor light go off, and then back on showing the last image, but in a non-responsive way, not accepting any input besides force shutdown.

I'm willing to help debug this, and can provide any additional information, just tell me what is needed, and how to get it.

Specs: 
```
OS: Manjaro Linux x86_64
Host: MacBookAir7,2 1.0
Kernel: 5.9.9-arch1-1-macbook / also tested with 5.10.6-majaro 
WM: i3
CPU: Intel i5-5250U (4) @ 1.600GHz
GPU: Intel HD Graphics 6000
Memory: 1149MiB / 3834MiB
```

patjak, 2021-02-08:
Hi, the driver doesn't support suspend/hibernation when the camera is running (streaming data). Is the camera on or off when you hibernate?

theotheroracle, 2021-05-25:
macboook air late 2012 edition, same issue, although it's not intermittent, it happens every time

alexshpilkin, 2021-05-25:
@theotheroracle wrote:
> journalctl didn't show any helpful information, unsure how else to check logs

You can also inspect the kernel message ring buffer directly using _dmesg_(1) (for example, `dmesg -w` if you want online output, like `journalctl -f` provides). Normally _systemd-journald_(8) picks up those messages and you’ll see them in the _journalctl_(1) output as well, but the full journal is frequently cluttered, so _dmesg_ can be helpful when you want to watch the kernel specifically.

theotheroracle, 2021-05-25:
so i suspended it, it froze as expected, and then i rebooted to check the logs, and it doesn't seem like there was any output since before the boot . i did copy the logs that mention facetimehd to a bin here :
https://bin.disroot.org/?da9704162eb72a9b#HRR1PZGx8SPPcSzeivsnFArpvYuuUSnzHhs2KtKSpNSF

theotheroracle, 2021-05-27:
should note that it's not just when the lid is closed, it happens every time the system suspends . regardless of wether the camera is open or ever has been opened since boot .

CasperVM, 2025-11-03:
I know this is an old issue, but posting it here nonetheless since I ran into this again today. You can work around it by writing a small script that unloads and loads the driver for suspend:

`sudo nano /usr/lib/systemd/system-sleep/fix-macbook-wakeup`

This also brings properly shuts the cpu cores down and brings them back up, as I've had issues in the past with that as well:

```bash
#!/bin/sh

switch_cpu () {
    for cpu in $(ls /sys/devices/system/cpu | egrep -i 'cpu[1-9][0-9]?'); do
      echo $1 | sudo tee /sys/devices/system/cpu/$cpu/online > /dev/null;
    done
}

echo "fix macbook wakeup"
case "$1/$2" in
  pre/*)
    echo "going to $2..."
    switch_cpu 0
    /usr/bin/rmmod facetimehd
    ;;
  post/*)
    echo "waking up from $2..."
    switch_cpu 1
    /usr/bin/modprobe facetimehd
   ;;
esac
```

Then just;
`sudo systemctl daemon-reload`

And you should be good..

Not a proper fix, but better than having to choose between suspend working, and having a camera.