patjak/facetimehd #242: modprobe: FATAL: Module bdc_pci not found.

schulzalbrecht, 2021-05-09:
I'm using Ubuntu 20.04 and a macbook pro 2015. 

when runnning the installation script on ubuntu (section "Ubuntu installation script" in the wiki page), the script throws an error when performing: 
`$ sudo modprobe -r bdc_pci`

The error is:
modprobe: FATAL: Module bdc_pci not found.

I've already tried to reboot the machine but it doens't help.

lucasmsoares96, 2021-05-15:
the -r flag in the modprobe command removes the module from the kernel. If you cannot remove it, it is because you do not have it. It's not a problem.

schulzalbrecht, 2021-05-15:
true. in the meantime the penny dropped and became clear to me that the command was actually just trying to remove something I didn't have. So I carried on executing the script and my camera is working now! And anyway thanks for replying to it, it can be helpful to other people in the future.

closed 2021-05-15 by schulzalbrecht