patjak/facetimehd #243: Resolution problem

antoniomokuzai, 2021-05-09:
Thanks for the nice work, at least you made the camera work, but I have a problem with camera resolution. 

According to the spec, FaceTime HD's physical max resolution is 5 MPixels, which translates to 2592 x 1944 pixels, but after I load the driver to make /dev/video0 show up, checking the resolution with the following command: 

v4l2-ctl --list-formats-ext -d /dev/video0

give me this output:

ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture
	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YVYU' (YVYU 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)

And this low quality result (1280 x 720) is consistent with quality of picture that is taken with this camera using your driver, I am just wondering if that is a built-in limitation of the driver or my configuration problem, thanks!

patjak, 2021-06-09:
Hi, the FacetimeHD cameras are only 720p (unless you have a newer model I haven't seen before). What type of MacBook do you have and where does it say it should support 5MPixels?

vrilutza, 2026-10-16:
Master lists every size the driver can scale to now (#331), from 320x240 up to what the sensor reports (1296x736 on my MacBookPro14,1), instead of a single 1280x720. No 5 MP mode though, that's the sensor.