patjak/facetimehd #244: Problems installing bcwc_pcie

random-integer, 2021-05-22:
Hi, I'm trying to install the drivers for the MacBook webcam. I'm following the instructions [here](https://github.com/patjak/bcwc_pcie/wiki/Installation#get-started-on-arch) to install the kernel module manually because I can't find "another one for the firmware itself". Here's what I did.

```
$ git clone https://github.com/patjak/facetimehd-firmware.git
...
$ cd facetimehd-firmware
$ make
... (just like https://github.com/patjak/bcwc_pcie/wiki/Get-Started#firmware-extraction )
$ sudo make install
... (just like https://github.com/patjak/bcwc_pcie/wiki/Get-Started#firmware-extraction )
$ cd ..
$ git clone https://github.com/patjak/bcwc_pcie.git
...
$ cd bcwc_pcie
$ make
make -C /lib/modules/5.10.36-2-MANJARO/build M=/home/randint/webcam_stuff/bcwc_pcie modules
make[1]: Entering directory '/usr/lib/modules/5.10.36-2-MANJARO/build'
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_ddr.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_hw.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_drv.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_isp.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_v4l2.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_buffer.o
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/fthd_debugfs.o
  LD [M]  /home/randint/webcam_stuff/bcwc_pcie/facetimehd.o
  MODPOST /home/randint/webcam_stuff/bcwc_pcie/Module.symvers
  CC [M]  /home/randint/webcam_stuff/bcwc_pcie/facetimehd.mod.o
  LD [M]  /home/randint/webcam_stuff/bcwc_pcie/facetimehd.ko
make[1]: Leaving directory '/usr/lib/modules/5.10.36-2-MANJARO/build'
$ sudo make install
make -C /lib/modules/5.10.36-2-MANJARO/build M=/home/randint/webcam_stuff/bcwc_pcie modules_install
make[1]: Entering directory '/usr/lib/modules/5.10.36-2-MANJARO/build'
  INSTALL /home/randint/webcam_stuff/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.10.36-2-MANJARO
make[1]: Leaving directory '/usr/lib/modules/5.10.36-2-MANJARO/build'
```

Here, I was wondering why it failed. Then, I read [this](https://github.com/patjak/bcwc_pcie/wiki#known-issues). I figured that I hadn't reboot before running `sudo make install` in `bcwc_pcie`, so I did, then tried again.

```
(in bcwc_pcie)
$ sudo make install
```

I got the same output again! What should I do?

Info:

```
OS: Manjaro Linux x86_64
Host: MacBookPro12,1 1.0
Kernel: 5.10.36-2-MANJARO
```

By the way, I have successfully installed this driver once (I forgot how I did it). But after a kernel update, it broke, and only until now that I decided to make it work again.

Please do ask me if there's something else you need to know to solve my problem.

random-integer, 2021-06-20:
Seriously? The makers aren't gonna look into this?

alexshpilkin, 2021-06-20:
@random-integer This looks like you haven’t set up module signing properly, _i.e._ a problem that has nothing to do with this driver specifically, it’s just that you need to set up a build environment for any out-of-tree module to work. Getting module signing to work is unfortunately fairly distro-specific and sometimes even specific to the UEFI Secure Boot firmware on your machine. It’s a mess, and a mess that is in large part due to Microsoft requiring Secure Boot from computer manufacturers for their “Windows Logo” certification. ... It’s too late to do anything about it, I guess.

I searched for `pem` in the project issues and ended up with #83, #89, #99, #107, #134, #160, #174, #176, #181, #191, #195, #210, and #238. To be fair, this _is_ a place where the documentation is lacking, guessing the correct search term requires at least a slight idea of what’s wrong, and not all of these links have useful discussion behind them, but generally people are going to be more responsive if your issue report indicates that you have looked through the bug tracker before posting.

If you want to get things to work the way you’ve been trying to, you’ll have to look into setting up module signing; I’ve tried to gather a collection of links the last time it came up, at https://github.com/patjak/bcwc_pcie/issues/238#issuecomment-771130767. Alternatively, given that you’re on Manjaro, have you considered using the [bcwc-pcie-git package](https://aur.archlinux.org/packages/bcwc-pcie-git) from AUR? It uses [DKMS](https://wiki.archlinux.org/title/Dynamic_Kernel_Module_Support) and should take care of these (and other) issues for you. You might also want to get [facetimehd-firmware](https://aur.archlinux.org/packages/facetimehd-firmware) to improve sensor calibration, although that requires wrangling some files out of macOS.

fultonm, 2021-08-11:
See if this answer can help you: https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules/1322832#1322832

random-integer, 2021-08-13:
> Alternatively, given that you’re on Manjaro, have you considered using the [bcwc-pcie-git package](https://aur.archlinux.org/packages/bcwc-pcie-git) from AUR? It uses [...]

@alexshpilkin I just tried out the AUR package, and it works! It didn't work a few months ago. Thanks!

closed 2021-08-13 by random-integer