patjak/facetimehd #245: License

Dekedro, 2021-07-08:
Currently this project has no licensing terms attached to it. While one could assume project is (legally) safe to use, it would be nice to have license clarifying that.

WhyNotHugo, 2021-07-20:
Different files mention GPLv2, so I assume that applies to the project as a whole too.

It's also based on existing GPLv2 code, so I think that's really the only choice the authors have anyway.

Dekedro, 2021-07-20:
Yes, looks like all source files contain headers with licensing terms. Seems I missed that.

Anyway, it would be useful to have full license text in a separate LICENSE file, so GitHub can detect and display license type.

patjak, 2021-08-10:
Yes, the license for the project is GPLv2. Previously you could choose a license for projects on github so that it would show on the project page. Looks like they have removed that feature.

patjak, 2021-08-11:
I've now changed the license text to SPDX license identifiers and included GPLv2 in a LICENSE file. Closing the issue.

closed 2021-08-11 by patjak