patjak/facetimehd #246: fails to install on manjaro

MightyPrince1, 2021-07-09:
when I tried to install it, it reported SSL errors

make -C /lib/modules/5.4.129-2-MANJARO/build M=/home/maximilian/bcwc_pcie modules_install
make[1]: Entering directory '/usr/lib/modules/5.4.129-2-MANJARO/build'
  INSTALL /home/maximilian/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.129-2-MANJARO
make[1]: Leaving directory '/usr/lib/modules/5.4.129-2-MANJARO/build'

MightyPrince1, 2021-07-09:
just to be perfectly clear the camera does not work

alexshpilkin, 2021-07-09:
OK, this is getting silly. TL;DR: this is a Secure Boot issue that has nothing to do with this specific driver. Bibliography and references: https://github.com/patjak/bcwc_pcie/issues/244#issuecomment-864510459. Easiest solution for Manjaro (probably, I don’t run Manjaro and never have): grab [bcwc-pcie-git from AUR](https://aur.archlinux.org/packages/bcwc-pcie-git), it uses DKMS which should solve this for you.

Can someone _please_ write up a list of instructions for people on different distros? I’m not hitting this issue personally and don’t trust myself to do it blindly.