patjak/facetimehd #247: Kernel config options?

fultonm, 2021-08-16:
When building this out-of-tree module using a kernel and headers built from source, which are the ideal kernel config options to choose so that this driver can build?

patjak, 2021-08-17:
lsmod should be able to tell you the dependencies of the facetimehd module.
```lsmod | grep facetimehd```

fultonm, 2021-08-17:
How can we find which Configuration Options do these correspond to when configuration with `make menuconfig` on the kernel?

When I `lsmod | grep facetimehd` this I see the following listed:
`videobuf2_dma_sg`
`videobuf2_v4l2`
`videobuf2_common`
`videodev`

Thank you for your response to this issue. I know this is not an issue with this driver but a user problem created by newbie hackers like me, but for MacBook users trying to experiment with linux development this may be the first thing they try to get their machine working, so I hope it helps someone in the future. I am happy to write some information in the Wiki after I fully understand.

patjak, 2021-08-24:
Hi, sorry for the delay. I don't know of a good way to do this. You can try searching in "menuconfig" by pressing '/' (forward slash), but when I tried finding the configs you're looking for I only found ways to enable it by enabling drivers that use them. One alternative to try would be to edit the .config file manually.