patjak/facetimehd #248: bcwc-pcie fails to build with latest GCC

ohell, 2021-08-20:
webcam stopped working after updating to kernel 5.9.9. So I tried to do a fresh install of bcwc-pcie-git. Get an error, only for the new kernel version:
`==> dkms install --no-depmod -m bcwc-pcie -v 0.5.7.r1.gf85a604 -k 4.19.204-1-MANJARO
==> dkms install --no-depmod -m bcwc-pcie -v 0.5.7.r1.gf85a604 -k 5.9.9-arch1-1-macbook
Error! Bad return status for module build on kernel: 5.9.9-arch1-1-macbook (x86_64)
Consult /var/lib/dkms/bcwc-pcie/0.5.7.r1.gf85a604/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m bcwc-pcie -v 0.5.7.r1.gf85a604 -k 5.9.9-arch1-1-macbook' returned 10
`
Looking at the log file, it seems package needs to be updated for newer libraries included with gcc? Also, I would be grateful if you can suggest a workaround (i.e. iis to possible to build using clang? what commands would be needed?)

`make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:283: /var/lib/dkms/bcwc-pcie/0.5.7.r1.gf85a604/build/fthd_hw.o] Error 1
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[1]: *** [scripts/Makefile.build:283: /var/lib/dkms/bcwc-pcie/0.5.7.r1.gf85a604/build/fthd_drv.o] Error 1
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[1]: *** [scripts/Makefile.build:283: /var/lib/dkms/bcwc-pcie/0.5.7.r1.gf85a604/build/fthd_ringbuf.o] Error 1
make: *** [Makefile:1784: /var/lib/dkms/bcwc-pcie/0.5.7.r1.gf85a604/build] Error 2
make: Leaving directory '/usr/lib/modules/5.9.9-arch1-1-macbook/build'
`

alexshpilkin, 2021-08-20:
Not the author, but it seems unlikely to me that this is a problem with this driver. These GCC messages indicate that you have link-time optimization (LTO) on, but some of the objects in the link have been by a different version of GCC than you have installed. Try nuking your build directories, maybe? (Why are you trying to build a kernel module with LTO anyway? Even for normal packages that’s not necessarily a good idea, and I don’t think the default makepkg configuration does it.)

ohell, 2021-08-20:
I did a clean build after the first time install failed. The errors are from that process. and I have not set anything explicitly about LTO. I just run `yay bcwc-pcie` and choose the option for clean build when asked.