patjak/facetimehd #249: Doesn't work in Arch VM of M1 Macbook Air

DUOLabs333, 2021-08-21:
I followed the instructions, yet my situation hasn't improved -- qemu still shows a green screen instead of my camera, even when I passed in the proper options `-device qemu-xhci -device usb-host,vendorid=0x106B,productid=0x000B`.

patjak, 2021-08-24:
I have no idea what camera is in the M1 machines. Is it even compatible? This driver is only for PCIe devices and not USB. Are you using PCI passthrough to make the device visible inside the virtual machine?

DUOLabs333, 2021-08-24:
For the first part, yes, for M1, Apple uses PCI camera. Two, i'm not sure how would I do that in ARM (all instructions online are for x64).

patjak, 2021-08-24:
Sorry I have no experience with QEMU on M1. Is the device listed when you run lspci inside the virtual machine?

DUOLabs333, 2021-08-24:
No, there's a host bridge, audio device, ethernet controller, display controller, keyboard controller, mouse controller, USB controller, and a SCSI storage controller.

zetaomegagon, 2021-12-22:
@DUOLabs333 as a baseline, you are going to need to figure out how to pass through PCIe from your host M1 Mac to Qemu. As for the issues still existing after that, based on @patjak 's knowledge of the hardware-- I'd say that ultimately this would be a feature request.

DUOLabs333, 2021-12-22:
Yeah, I still haven't figured out how to pass in that or the microphone.

closed 2024-09-10 by DUOLabs333