patjak/facetimehd #250: On Linux Mint permission (group) problems

danjde, 2021-09-06:
Hi patjak,
I've installed facetimehd module and loaded it generate rightly the device /dev/video0.

This device has the following permission:

ls -l /dev/video*
crw-rw----+ 1 root video 81, 0 set  2 23:32 /dev/video0

But the normal user are not member of "video" group.

With the following command we can add the user to the "video" group and solve the issue:
 
'sudo usermod -a -G video $LOGNAME'

I still have to check the procedure physically on the computer (an AirBook of 2015) if it solves I let you know

I think should be fine adding this scenario on your documentation.

Thanks!

danjde, 2021-09-06:
I've write simple (and tested on Linux Mint 20.2 ) script, for creating and installing the module and build the dkms module -> 

```
#!/bin/bash

sudo apt install linux-headers-$(uname -r) git kmod libssl-dev xz-utils curl cpio
cd /tmp
mkdir facetimehd
cd facetimehd/
git clone https://github.com/patjak/facetimehd-firmware.git
cd facetimehd-firmware
make
sudo make install
cd ../
git clone https://github.com/patjak/bcwc_pcie.git
cd bcwc_pcie
make
sudo make install
sudo depmod
sudo modprobe facetimehd
# rm -Rf /tmp/facetimehd/
# genero il pacchetto DKMS...
echo -e "GENERAZIONE \e[92mDEL PACCHETTO DKMS IN CORSO..."
sudo apt-get install build-essential
sudo apt install -y debhelper dkms
sudo mkdir /usr/src/facetimehd-0.1
sudo dpkg -r bcwc-pcie
sudo cp -r * /usr/src/facetimehd-0.1/
cd /usr/src/facetimehd-0.1/
sudo rm backup-*tgz bcwc-pcie_*deb
sudo make clean
sudo dkms add -m facetimehd -v 0.1
sudo dkms build -m facetimehd -v 0.1
sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
sudo dkms mkdeb -m facetimehd -v 0.1 --source-only
sudo cp /var/lib/dkms/facetimehd/0.1/deb/facetimehd-dkms_0.1_all.deb /root/
sudo rm -r /var/lib/dkms/facetimehd/
sudo dpkg -i /root/facetimehd-dkms_0.1_all.deb
echo -e "PROCEDURA COMPLETATA \e[92mRIAVVIARE IL COMPUTER!"
```

patjak, 2021-09-08:
Hi, everyone can edit the wiki so feel free to add the information if needed. This particular issue is distro specific so try placing it somewhere where it makes sense. Thanks