patjak/facetimehd #252: modprobe -r bdc_pci output: modprobe: FATAL : Module bdc_pci not found. (zorin os 16)

gamerseye123, 2021-09-19:
so im trying to get the driver to load on startup, and i have to run `modprobe -r bdc_pci` but it cant find  bdc_pcie. what do i do?

zetaomegagon, 2021-12-22:
I'm not sure I understand. That `modprobe -r bdc_pci` doesn't find `bdc_pcie` seems to imply that it isn't loaded.

Why do you need `modeprobe -r bdc_pci` to work?

zetaomegagon, 2022-01-03:
@gamerseye123 sorry for the late reply. We'll need more context to troubleshoot your issue. For instance...

1. What steps did you use to install the driver?
2. What steps did you take to attempt loading the driver at boot?
3. Provide the logs from the following command, or similar, just after booting up:
`journalctl -x | grep -iE 'facetime(|hd| hd)`

zetaomegagon, 2022-01-03:
I also suggest changing the title to describe the current state of the issue. Something like: `[YourOperatingSystemHere] - Unable to get driver to load at boot`

patjak, 2022-01-13:
bdc_pcie is removed from more recent kernels so it is no longer required to unload it. If it fails to unload, then it's simply not loaded :)
Automatic loading of the facetimehd module on startup should be fixed with the latest commit 8431504df5a1d058fc5888f6bb695de859dbf22f

pasan93, 2022-05-05:
> so im trying to get the driver to automatically load on startup, and i have to run `modprobe -r bdc_pci` but it cant find bdc_pcie. what do i do?

Same error on Pop!_OS 21.10 and 22.04

larsblumberg, 2022-12-19:
Just remove (=do not execute) the line `modprove -r bdc_pci`, it's no longer required for recent Ubuntu/Pop_OS! versions.