patjak/facetimehd #254: SSL Errrors while compiling DKMS module on Fedora

sxiii, 2021-11-01:
```
[root@fedora-mbp facetimehd-0.1]# dkms build -m facetimehd -v 0.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.12.11-300.mbp.fc33.x86_64 -C /lib/modules/5.12.11-300.mbp.fc33.x86_64/build M=/var/lib/dkms/facetimehd/0.1/build....
Signing module /var/lib/dkms/facetimehd/0.1/build/./facetimehd.ko

At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: /root/dkms.key: No such file or directory
Error! Bad exit status 1 for sign tool.
Consult /var/lib/dkms/facetimehd/0.1/5.12.11-300.mbp.fc33.x86_64/x86_64/log/make.log for more information.
```

OS: Fedora 34
Kernel: Linux fedora-mbp 5.12.11-300.mbp.fc33.x86_64

The webcam does not work. However it worked at some moment on Fedora and my MBP 16,1. So I guess it is possible to fix...

gdelfino, 2021-11-17:
I am getting a very similar error under Ubuntu (elementary OS 6 Odin):

```
gdelfino@MBP2015:~/bcwc_pcie$ make
make -C /lib/modules/5.11.0-40-generic/build M=/home/gdelfino/bcwc_pcie modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-40-generic'
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_ddr.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_hw.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_drv.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_isp.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_v4l2.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_buffer.o
  CC [M]  /home/gdelfino/bcwc_pcie/fthd_debugfs.o
  LD [M]  /home/gdelfino/bcwc_pcie/facetimehd.o
  MODPOST /home/gdelfino/bcwc_pcie/Module.symvers
  CC [M]  /home/gdelfino/bcwc_pcie/facetimehd.mod.o
  LD [M]  /home/gdelfino/bcwc_pcie/facetimehd.ko
  BTF [M] /home/gdelfino/bcwc_pcie/facetimehd.ko
Skipping BTF generation for /home/gdelfino/bcwc_pcie/facetimehd.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-40-generic'

gdelfino@MBP2015:~/bcwc_pcie$ sudo make install
make -C /lib/modules/5.11.0-40-generic/build M=/home/gdelfino/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-40-generic'
  INSTALL /home/gdelfino/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.11.0-40-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-40-generic'

gdelfino@MBP2015:~/bcwc_pcie$ uname -a
Linux MBP2015 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

```

jucajuca, 2021-11-22:
I tried this and it solved the problem with SSL, but still no luck in installing the driver:

cd /lib/modules/$(uname -r)/build/certs

sudo tee x509.genkey > /dev/null << 'EOF'
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts
[ req_distinguished_name ]
CN = Modules
[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid
EOF

sudo openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem

jaceygan, 2021-12-14:
@gdelfino did you get to fix your issue? I get the same exact error (shown below). The webcam does work after running this. However, cheese will not be able to detect the webcam again after a reboot and I have to run bash script.sh again.

`acey@jacey-MBPU:~$ bash script.sh
Compiling firmware
make: *** No targets specified and no makefile found.  Stop.
done

Installing firmware
[sudo] password for jacey: 
make: *** No rule to make target 'install'.  Stop.
done

Compiling driver
make -C /lib/modules/5.11.0-41-generic/build M=/home/jacey/bcwc_pcie modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
  CC [M]  /home/jacey/bcwc_pcie/fthd_ddr.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_hw.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_drv.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_isp.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_v4l2.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_buffer.o
  CC [M]  /home/jacey/bcwc_pcie/fthd_debugfs.o
  LD [M]  /home/jacey/bcwc_pcie/facetimehd.o
  MODPOST /home/jacey/bcwc_pcie/Module.symvers
  CC [M]  /home/jacey/bcwc_pcie/facetimehd.mod.o
  LD [M]  /home/jacey/bcwc_pcie/facetimehd.ko
  BTF [M] /home/jacey/bcwc_pcie/facetimehd.ko
Skipping BTF generation for /home/jacey/bcwc_pcie/facetimehd.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-41-generic'
done

Installing driver
make -C /lib/modules/5.11.0-41-generic/build M=/home/jacey/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
  INSTALL /home/jacey/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.11.0-41-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-41-generic'
done

Running depmod
done

Running modprobe -r bdc_pci
modprobe: FATAL: Module bdc_pci not found.
done

Loading driver
done

jacey@jacey-MBPU:~$ dmesg | grep facetimehd
[  286.004852] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  286.005021] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  286.009894] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  286.009940] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  286.019874] facetimehd 0000:02:00.0: PLL reset finished
[  286.019876] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  286.019889] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  286.029815] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  286.049693] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  286.049709] facetimehd 0000:02:00.0: STRAP valid
[  286.049710] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  286.049717] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  286.049726] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  286.049736] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  286.049737] facetimehd 0000:02:00.0: Using step size 151
[  286.049741] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[  286.049746] facetimehd 0000:02:00.0: Virtual VTT enabled
[  286.070101] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  286.070115] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  286.070309] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  286.188060] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[  286.242141] facetimehd 0000:02:00.0: ISP woke up after 0ms
[  286.242156] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[  286.242193] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[  286.252149] facetimehd 0000:02:00.0: ISP second int after 0ms
[  286.252153] facetimehd 0000:02:00.0: Channel description table at 00800000
[  286.262328] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[  286.262332] facetimehd 0000:02:00.0: Enabling interrupts
jacey@jacey-MBPU:~$ uname -a
Linux jacey-MBPU 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
`

zetaomegagon, 2021-12-22:
@jaceygan -- I did what @jucajuca did by creating a `signing_key.pem` then followed the instructions for installing as a dkms module in Fedora.

I'm on Fedora 35: `5.15.10-200.fc35.x86_64`.

A tiny bit off topic, but you probably shouldn't be running the script over again just to manually load the kernel module-- can't you do `sudo modprobe facetimehd`? When you re-run the script you are re-compiling the kernel too...

Can you reboot and just see if `sudo modprobe facetimehd` works... It won't fix the issue, but as a baseline, we should know if just running `modprobe` works.

As an aside-- if `modprobe` does work, you could write a systemd unit to load the kernel module sometime after boot

zetaomegagon, 2021-12-22:
@jaceygan also just realized that your `dmesg` output is from *after* you loaded the module?

What about just after boot...or running something like `journalctl -x | grep -iE 'facetime(|hd| hd)'`

jaceygan, 2021-12-22:
@zetaomegagon thanks for this. 

I had to get my Zoom (with webcam) up and running and had no time to debug further. I went to win10 on bootcamp and everything works out of the box.

zetaomegagon, 2021-12-22:
@jaceygan I hear you. If I had had a choice, I'd have done something similar, but I had no option but to get the FaceTime HD cam working in Fedora (or some flavor of linux). I'm super happy I've got the driver working-- but what a drag. Why did Apple have to be so "different"?

jaceygan, 2021-12-24:
@zetaomegagon i had some time today given its the holiday season.

Instead of installing Ubuntu, I installed Pop_OS!. And I followed the instructions for Debian instead of Ubuntu.

My webcam works great now! It works after a reboot too!

zetaomegagon, 2021-12-25:
Glad to hear it @jaceygan !
--
Best,
  Elijah Beale

Sent via FairEmail[https://email.faircode.eu/]

pasan93, 2022-05-05:
> Hey patjak, I did get my web cam working, though I'd been running Fedora 34 or 35. If you are getting the SSL error: I think there is another SSL or Cert issue on this repo with directions for generating the missing cert. It's something like that. I don't remember what I did, honestly-- it feels like a lifetime since I had to fix it. I can try to take a look around on Sunday and post back. I have an infant in one arm right now :) May 5, 2022 10:00:15 Pasan ***@***.***>:
> […](#)

Thanks for the quick reply mate.
I tried Ubuntu 21.10, 22.04, Pop!_OS 21.10, 22.04, Fedora 35. No luck at all. (Now I'm kind of tired after testing and distro hopping LOL). 

It will be great if you can let me know how to fix it.
Family comes first mate👶, I can wait till Sunday. 👍🏽

zetaomegagon, 2022-05-05:
Hey Pasan,

If you are going to try Fedora 35 again, I dug up my comment on the copr repo for the driver. Might give you some clues, though my process isn't explicitly stated:

https://discussion.fedoraproject.org/t/frgt10-facetimehd-dkms/19736/2

pasan93, 2022-05-06:
> Hey Pasan, If you are going to try Fedora 35 again, I dug up my comment on the copr repo for the driver. Might give you some clues, though my process isn't explicitly stated: https://discussion.fedoraproject.org/t/frgt10-facetimehd-dkms/19736/2 May 5, 2022 10:38:45 Pasan ***@***.***>:
> […](#)

Thanks, I'll give it a try.