patjak/facetimehd #256: Ubuntu 21.10 Crash

GMAzrael, 2021-11-08:
Upgraded from 21.04 to 21.10 with PopOS and it crashed on the firmware loading. https://gist.github.com/GMAzrael/ba85549089489bddbd992889a0804d50

usernamefoo, 2021-11-08:
Can confirm with a new install of PopOS 21.10 beta2 on a 2015 macbook pro. This module seems to not like linux 5.15.

GMAzrael, 2021-11-29:
@patjak any luck investigating this one?

patjak, 2021-11-29:
It seems the Apple ACPI is doing something wrong. Did you also update to the latest firmware version in MacOS (normal MacOs updates) or just PopOS?

GMAzrael, 2021-11-29:
Didnt do any updates to MacOS for some time. My 2012 MBP wouldn't get Monterey. I didn't think my camera got a firmware update in some time.

sylqvq, 2022-01-04:
I'm not 100% sure if this contributes, but considering that building fails on my Pop 21.10 install (MBA 7,1), I could only assume.

`Skipping BTF generation for /home/syl/bcwc_pcie/facetimehd.ko due to unavailability of vmlinux`

Building on this exact laptop had been previously flawless; even on latest Apple firmware.

ivanperez-keera, 2022-03-03:
I get the same `Skipping BTF generation for /home/syl/bcwc_pcie/facetimehd.ko due to unavailability of vmlinux` notice. Could this be the reason?

sylqvq, 2022-03-23:
I did do a little research a good while back when I still used my MacBook, and all I could find is other DKMS modules failing to install because of an updated kernel. There's no "one size fits all" solution for this error in particular; every result I found had a different solution for their issue mentioned.

satmandu, 2022-04-04:
I was able to build a dkms package which worked just fine in 22.04 with kernel 5.17.1.

First install the 5.1.7.1 kernel from here: https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.17.1/
Then build and install the dkms package and the firmware files:
```
# Install required apps: 
sudo apt install debhelper git unrar
# get source
git clone https://github.com/patjak/bcwc_pcie.git
cd bcwc_pcie/
sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
sudo dkms mkdeb -m facetimehd -v 0.1 --source-only
# Now install the deb package in /var/lib/dkms/facetimehd/0.1/deb
sudo dpkg -i /var/lib/dkms/facetimehd/0.1/deb/*.deb
cd /tmp
curl -OLf https://download.info.apple.com/Mac_OS_X/031-30890-20150812-ea191174-4130-11e5-a125-930911ba098f/bootcamp5.1.5769.zip
unzip bootcamp5.1.5769.zip
cd BootCamp/Drivers/Apple/
unrar x AppleCamera64.exe
dd bs=1 skip=1663920 count=33060 if=AppleCamera.sys of=9112_01XX.dat
dd bs=1 skip=1644880 count=19040 if=AppleCamera.sys of=1771_01XX.dat
dd bs=1 skip=1606800 count=19040 if=AppleCamera.sys of=1871_01XX.dat
dd bs=1 skip=1625840 count=19040 if=AppleCamera.sys of=1874_01XX.dat
sudo cp *.dat /lib/firmware/facetimehd/
# and then reboot
```

ivanperez-keera, 2022-05-10:
Do you need that specific kernel? Could it work with: 5.13.0-40-generic?

ivanperez-keera, 2022-05-10:
I am getting:

```
$ sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
Error! DKMS tree does not contain: facetimehd-0.1
Build cannot continue without the proper tree.
```

ivanperez-keera, 2022-05-10:
I was just able to install it with:
```
# make
# make install
```

Using `checkinstall` did not work for me.  It created a deb without the kernel module (i.e., `*.ko` file).

I'd prefer to automate this process. Is there a chance that we could have 1) a PPA for this module? and 2) a way for the module to be re-compiled when we install a new kernel?

satmandu, 2022-05-10:
Try 
```
git clone https://github.com/patjak/bcwc_pcie.git
cd bcwc_pcie/
sudo mkdir /usr/src/facetimehd-0.1
cp -r * /usr/src/facetimehd-0.1/
dkms add -m facetimehd -v 0.1
sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
sudo dkms mkdeb -m facetimehd -v 0.1 --source-only
```

networkdude2k, 2023-01-04:
I realize this thread is a few months old.  I am currently trying to run the dkms mkdsc and the dkms mkdeb steps.  It appears that the mkdsc and mkdeb actions have been removed from dkms on Ubuntu 22.10.  I confess to being somewhat of a noob when it comes to compiling drivers.  All the steps have run smoothly up to now, and I have just kind of hit a wall. Any help would be much appreciated.  When I run apt to install dkms it gives me:

networkdude@MacBookPro:~$ sudo apt install dkms
[sudo] password for networkdude: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (3.0.6-2ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

When I run dkms alone to view the options, I get this (note that mkdsc & mkdeb are missing):
Usage: /usr/sbin/dkms [action] [options]
  [action]  = { add | remove | build | install | uninstall | match | autoinstall |
                mktarball | ldtarball | status }
  [options] = [-m module] [-v module-version] [-k kernel-version] [-a arch]
              [-c dkms.conf-location] [-q] [--force] [--force-version-override] [--all]
              [--templatekernel=kernel] [--directive='cli-directive=cli-value']
              [--config=kernel-.config-location] [--archive=tarball-location]
              [--kernelsourcedir=source-location]
              [--binaries-only] [--source-only] [--verbose]
              [--no-depmod] [--modprobe-on-install] [-j number] [--version]

Ubuntu 22.10 on a 2017 MacBook Pro 15" (14,3), kernel version 5.19.0-26-generic.

jorge-vitrubio, 2023-10-11:
> I am getting:
> 
> ```
> $ sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
> Error! DKMS tree does not contain: facetimehd-0.1
> Build cannot continue without the proper tree.
> ```

the howto has to be updated, it seem `dkms`  in verson 3.x has dropped support for distro specific packagin

https://github.com/dell/dkms/commit/68b083eaa3f71c166adfece8e4f760e0cdf96185

so this man does not work any more in debian stable, testing and unstable

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1009179

lbtbs, 2023-10-22:
> Try
> 
> ```
> git clone https://github.com/patjak/bcwc_pcie.git
> cd bcwc_pcie/
> sudo mkdir /usr/src/facetimehd-0.1
> sudo cp -r * /usr/src/facetimehd-0.1/
> sudo dkms add -m facetimehd -v 0.1
> sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
> sudo dkms mkdeb -m facetimehd -v 0.1 --source-only
> ```

I tried and got this error:

sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
Error! Unknown action specified: ""
Usage: /usr/sbin/dkms [action] [options]
  [action]  = { add | remove | build | install | uninstall | match | autoinstall |
                mktarball | ldtarball | status }
  [options] = [-m module] [-v module-version] [-k kernel-version] [-a arch]
              [-c dkms.conf-location] [-q] [--force] [--force-version-override] [--all]
              [--templatekernel=kernel] [--directive='cli-directive=cli-value']
              [--config=kernel-.config-location] [--archive=tarball-location]
              [--kernelsourcedir=source-location]
              [--binaries-only] [--source-only] [--verbose]
              [--no-depmod] [--modprobe-on-install] [-j number] [--version]

bagong, 2023-10-22:
I can confirm this last one, but I doubt it matters... Unfortunately this repo has been renamed and only half refactored, so working through it is quite confusing... The driver works however, right. It's quite something to run a 2014 wrbcam with a 6.5 kernel (Ubuntu 23.10)

CaptainMorgan12, 2024-02-11:
I have Ubuntu 23 and Kernel 6.5 so is there a chance it works? The facetimehd camera is in my thunderbolt display not the macbook pro.

ryosuketc, 2024-05-05:
By any chance has anyone ever worked this out?  I thought about downgrading `dkms` to 2.x (maybe `2.8.4-3`: https://tracker.debian.org/pkg/dkms) but it's not found.
The Kernel is `6.8.0-31-generic` on `Ubuntu 24.04 LTS`.

```
$ apt list dkms -a
Listing... Done
dkms/noble 3.0.11-1ubuntu13 all
```

> > I am getting:
> > ```
> > $ sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
> > Error! DKMS tree does not contain: facetimehd-0.1
> > Build cannot continue without the proper tree.
> > ```
> 
> the howto has to be updated, it seem `dkms` in verson 3.x has dropped support for distro specific packagin
> 
> [dell/dkms@68b083e](https://github.com/dell/dkms/commit/68b083eaa3f71c166adfece8e4f760e0cdf96185)
> 
> so this man does not work any more in debian stable, testing and unstable
> 
> https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1009179