patjak/facetimehd #257: Fedora Silverblue

yavko, 2022-01-20:
Fedora silverblue uses rpm-ostree instead of dnf and works differently, can someone please help me out on how i can set up the driver for silverblue. Also silverblue has an immutable filesystem.

moraitis, 2024-10-24:
@yavko , I found this copr repo with instructions that worked for my MacBook Pro (Retina, 15-inch, Mid 2014), 2.8 GHz Intel Core i7, MacBookPro11,2 running Fedora Silverblue 40.

The instructions were here:
- https://copr.fedorainfracloud.org/coprs/mulderje/intel-mac-rpms 

and here:
-  https://discussion.fedoraproject.org/t/mulderje-intel-mac-rpms/130045

Two simple commands that get your facetime camera working great in Silverblue.

1. install copr repo

```bash
sudo curl -L -o /etc/yum.repos.d/_copr_mulderje-intel-mac-rpms.repo \
    "https://copr.fedorainfracloud.org/coprs/mulderje/intel-mac-rpms/repo/fedora-$(rpm -E %fedora)/mulderje-intel-mac-rpms-fedora-$(rpm -E %fedora).repo"
```

2. now install facetime

```bash
rpm-ostree install facetimehd-kmod
```

yavko, 2024-10-27:
Yeah I think this works

closed 2024-10-27 by yavko