patjak/facetimehd #259: facetimehd: probe of 0000:04:00.0 failed with error -5

aliris-schmidt, 2022-02-01:
Followed installation instruction from Wiki and from [this site](https://gist.github.com/johnjeffers/3006011ec7767a4101cdd118e8d64290#facetime-camera) before. Always worked very nice. Now it stopped working and reinstall does not help either.

Ubuntu 21.10 
5.13.0-28-generic

[    1.868915] facetimehd: loading out-of-tree module taints kernel.
[    1.868970] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[    1.869590] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
[    1.869787] facetimehd 0000:04:00.0: Setting 64bit DMA mask
[    1.875572] facetimehd 0000:04:00.0: S2 PCIe link init succeeded
[    1.875603] facetimehd 0000:04:00.0: Refclk: 25MHz (0xa)
[    1.885536] facetimehd 0000:04:00.0: PLL reset finished
[    1.885538] facetimehd 0000:04:00.0: Waiting for S2 PLL to lock at 450 MHz
[    1.885551] facetimehd 0000:04:00.0: S2 PLL is locked after 10 us
[    1.895477] facetimehd 0000:04:00.0: S2 PLL is in bypass mode
[    1.915356] facetimehd 0000:04:00.0: DDR40 PHY PLL locked on safe settings
[    1.915370] facetimehd 0000:04:00.0: STRAP valid
[    1.915371] facetimehd 0000:04:00.0: Configuring DDR PLLs for 450 MHz
[    1.915378] facetimehd 0000:04:00.0: DDR40 PLL is locked after 0 us
[    1.915389] facetimehd 0000:04:00.0: First DDR40 VDL calibration completed after 2 us
[    1.915400] facetimehd 0000:04:00.0: Second DDR40 VDL calibration completed after 2 us
[    1.915401] facetimehd 0000:04:00.0: Using step size 158
[    1.915406] facetimehd 0000:04:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[    1.915411] facetimehd 0000:04:00.0: Virtual VTT enabled
[    1.935769] facetimehd 0000:04:00.0: S2 DRAM memory address: 0x22159559
[    1.935797] facetimehd 0000:04:00.0: Rewrite DDR mode registers succeeded
[    1.936107] facetimehd 0000:04:00.0: Full memory verification succeeded! (0)
[    2.073811] facetimehd 0000:04:00.0: Loaded firmware, size: 1392kb
[   12.027408] facetimehd 0000:04:00.0: Init failed! No wake signal
[   12.027873] facetimehd: probe of 0000:04:00.0 failed with error -5

aliris-schmidt, 2022-02-05:
Tried it again on a clean install of Ubuntu 21.10 
5.13.0-28-generic
MacBookPro11,3

Made every step of Firmware extraction and then Install script but the camera won't work.

```
[    5.537878] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[    5.552886] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
[    5.553799] facetimehd 0000:04:00.0: Setting 64bit DMA mask
[    5.566299] facetimehd 0000:04:00.0: S2 PCIe link init succeeded
[    5.566363] facetimehd 0000:04:00.0: Refclk: 25MHz (0xa)
[    5.576310] facetimehd 0000:04:00.0: PLL reset finished
[    5.576316] facetimehd 0000:04:00.0: Waiting for S2 PLL to lock at 450 MHz
[    5.739092] facetimehd 0000:04:00.0: Failed to lock S2 PLL: 0xc902c902
[    5.759055] facetimehd 0000:04:00.0: DDR40 PHY PLL locked on safe settings
[    5.759087] facetimehd 0000:04:00.0: STRAP valid
[    5.759088] facetimehd 0000:04:00.0: Configuring DDR PLLs for 450 MHz
[    5.759106] facetimehd 0000:04:00.0: DDR40 PLL is locked after 0 us
[    5.759135] facetimehd 0000:04:00.0: First DDR40 VDL calibration completed after 0 us
[    5.759150] facetimehd 0000:04:00.0: Second DDR40 VDL calibration completed after 0 us
[    5.759151] facetimehd 0000:04:00.0: Using step size 159
[    5.759161] facetimehd 0000:04:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[    5.759170] facetimehd 0000:04:00.0: Virtual VTT enabled
[    5.779560] facetimehd 0000:04:00.0: S2 DRAM memory address: 0x22159559
[    5.779578] facetimehd 0000:04:00.0: Rewrite DDR mode registers succeeded
[    5.779782] facetimehd 0000:04:00.0: Full memory verification succeeded! (0)
[    5.897933] facetimehd 0000:04:00.0: Loaded firmware, size: 1392kb
[   15.853465] facetimehd 0000:04:00.0: Init failed! No wake signal
[   15.853742] facetimehd: probe of 0000:04:00.0 failed with error -5
```

aliris-schmidt, 2022-02-05:
I also get this error on sudo modprobe facetimehd.

`modprobe: FATAL: Module facetimhd not found in directory /lib/modules/5.13.0-28-generic`

Which I can't explain because as mentioned it is a new system no updates etc.

aliris-schmidt, 2022-02-07:
Just installed MacOS on an external drive to check if the camera works there. The camera is not recognized in the systems report. Dont know if that is an hardware error.

patjak, 2022-02-15:
Hi
All I can tell from the log is that the camera is not "booting" after the firmware is loaded. But as you say, if the camera is not working in MacOS then I suspect it's not an issue with our driver.

jplx, 2022-02-19:
I have used this driver successfully on Macbook Pro running Gentoo Linux. However, it failed recently because apparently EAPI 5 is not more supported. I tried to modify to EAPI=6 in the ebuild file but it fails the "source prepare" step. It seems that it does not even fetch and unpack when EAPI is set to 6.
Would it be possible for you to update both ebuild files to EAPI 6 or 7? The files are bcwc_pcie-9999-r1.ebuild and apple_facetimehd_firmware-1.43.0-r1.ebuild. Thank you in advance.

jplx, 2022-02-21:
I was able to make it work with EAPI7 and the attached ebuild. It requires to download de bcwc_pcie-master.zip file into the DISTDIR directory defined in /etc/portage/make.conf
It also requires to add the following FEATURES in the /etc/portage/make.conf
FEATURES="<your existing features> -sandbox -userpriv -usersandbox"
As I am new to ebuild, this looks like a hack but it works on my Macpro.
Your comments and corrections would be greatly appreciated.

[bcwc_pcie-9999-r2.ebuild.txt](https://github.com/patjak/facetimehd/files/8106459/bcwc_pcie-9999-r2.ebuild.txt)

patjak, 2022-03-10:
sorry, I don't know anything about ebuild. You need to ask whoever is maintaining the Gentoo package. I'm closing this issues since it seems to be about broken hardware.

closed 2022-03-10 by patjak

jplx, 2022-03-11:
OK. Thank you for letting me know.