patjak/facetimehd #260: No camera on MacBook Pro 2017 14,2

truonggiang0710, 2022-03-01:
Hello team,
I clone the driver and install it as per your instruction but the camera is not working.
Please help me.
My operator is Kali Linux 5.16.0-kali-amd64

truonggiang0710, 2022-03-10:
Hi @patjak , Do you have any ideas to resolve the issue?

patjak, 2022-03-10:
Hi. Please have a look at other issues to see if yours is already reported. Otherwise you need to describe your system and your problem in more detail.

truonggiang0710, 2022-03-10:
Hi men,
I just review all of the issues but the camera is still a problem.
My system is live persistent Kali Linux (5.16.0).
I use "lsusb" => Saw 05ac:8600 Apple, In. iBridge
Then "usb-devices" => I don't find the "05ac:8600"
What's the problem here? Please help me.