patjak/facetimehd #262: Make fails

ThamognyaKodi, 2022-03-15:
```sh
make -C /lib/modules/5.16.11-gentoo-x86_64/build M=/home/jeff/Documents/video-driver/facetimehd-0.5.7 modules
make[1]: Entering directory '/usr/src/linux-5.16.11-gentoo'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-pc-linux-gnu-gcc (Gentoo 11.2.0 p1) 11.2.0
  You are using:           gcc (Gentoo 11.2.0 p1) 11.2.0
  MODPOST /home/jeff/Documents/video-driver/facetimehd-0.5.7/Module.symvers
ERROR: modpost: "vb2_dma_sg_memops" [/home/jeff/Documents/video-driver/facetimehd-0.5.7/facetimehd.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:134: /home/jeff/Documents/video-driver/facetimehd-0.5.7/Module.symvers] Error 1
make[2]: *** Deleting file '/home/jeff/Documents/video-driver/facetimehd-0.5.7/Module.symvers'
make[1]: *** [Makefile:1761: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-5.16.11-gentoo'
make: *** [Makefile:12: all] Error 2

```

This is my error

I am running gentoo and am not too sure what went wrong.

I am using this guide rather than the outdated ebuild: 

https://gist.github.com/johnjeffers/3006011ec7767a4101cdd118e8d64290#facetime-camera

ricjcosme, 2022-06-08:
Having the same problem in RHEL 9, kernel 5.14.0-70.13.1.el9_0.x86_64
```
make -C /lib/modules/5.14.0-70.13.1.el9_0.x86_64/build M=/home/rc/Downloads/facetimehd modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-70.13.1.el9_0.x86_64'
  CC [M]  /home/rc/Downloads/facetimehd/fthd_ddr.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_hw.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_drv.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_ringbuf.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_isp.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_v4l2.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_buffer.o
  CC [M]  /home/rc/Downloads/facetimehd/fthd_debugfs.o
  LD [M]  /home/rc/Downloads/facetimehd/facetimehd.o
  MODPOST /home/rc/Downloads/facetimehd/Module.symvers
ERROR: modpost: "vb2_dma_sg_memops" [/home/rc/Downloads/facetimehd/facetimehd.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:150: /home/rc/Downloads/facetimehd/Module.symvers] Error 1
make[2]: *** Deleting file '/home/rc/Downloads/facetimehd/Module.symvers'
make[1]: *** [Makefile:1777: modules] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-70.13.1.el9_0.x86_64'
make: *** [Makefile:12: all] Error 2
```

GlenWalker2, 2023-02-08:
I have a slightly different kernel but the exact same problem.

`5.14.0-162.12.1.el9_1.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Dec 20 06:06:30 EST 2022 x86_64 x86_64 x86_64 GNU/Linux`

I'm also on RedHat (RHEL) 9.  Is it some quirk in the way our kernel has been built?  Will we have to build another kernel with different flags set?  Its been a long time since I've built a kernel so will have to refresh my memory a bit.

Have hunted around in the source code and on the web and cannot work out what is going on.  If I'm understanding the code correctly then the source file "fthd_v4l2.c" includes the header "media/videobuf2-dma-sg.h" which declares vb2_dma_sg_memops as:

`extern const struct vb2_mem_ops vb2_dma_sg_memops;`

and I can only assume that it hasn't been defined and included in the kernel as an available function.  What do you think?

GlenWalker2, 2023-02-09:
An update today and I'm going to keep looking but it appears my suspicions might be right about it not being enabled in the kernel.  This post is related: github.com/toaster/gentoo-overlay/issues/4

scheepers, 2023-05-22:
Here too, but on Ubuntu
```
make -C /lib/modules/5.19.0-42-generic/build M=/home/me/bcwc_pcie modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-42-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-42-generic'
```

ellandisan, 2024-01-11:
I have same situtation with  same error , do you have any resolution for this ?
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           
  CC [M]  /home/jakub/bcwc_pcie/fthd_ddr.o