patjak/facetimehd #267: No working on MacBook pro  mid 2015

mirfan899, 2022-06-04:
I have installed the driver but screen is black and green light near cam is on. 

Here is how its look.

[image]

Mykernal
```text
5.13.0-44-generic
```

lsmod result
[image]

It seems it does not support kernal 5.13

Here is further output of device
[image]

ledisthebest, 2022-06-07:
What's the output of `lspci -nnk` and please try it in OBS Studio

mirfan899, 2022-06-07:
Here is the output
```shell
lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation Crystal Well DRAM Controller [8086:0d04] (rev 08)
	DeviceName: SATA
	Subsystem: Apple Inc. Crystal Well DRAM Controller [106b:0147]
00:01.0 PCI bridge [0604]: Intel Corporation Crystal Well PCI Express x16 Controller [8086:0d01] (rev 08)
	Kernel driver in use: pcieport
00:01.1 PCI bridge [0604]: Intel Corporation Crystal Well PCI Express x8 Controller [8086:0d05] (rev 08)
	Kernel driver in use: pcieport
00:02.0 VGA compatible controller [0300]: Intel Corporation Crystal Well Integrated Graphics Controller [8086:0d26] (rev 08)
	Subsystem: Apple Inc. Crystal Well Integrated Graphics Controller [106b:0147]
	Kernel driver in use: i915
	Kernel modules: i915
00:03.0 Audio device [0403]: Intel Corporation Crystal Well HD Audio Controller [8086:0d0c] (rev 08)
	Subsystem: Apple Inc. Crystal Well HD Audio Controller [106b:0147]
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
	Subsystem: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:7270]
	Kernel driver in use: xhci_hcd
	Kernel modules: xhci_pci
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
	Subsystem: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller [8086:7270]
	Kernel driver in use: mei_me
	Kernel modules: mei_me
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
	Subsystem: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:7270]
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
	Kernel driver in use: pcieport
00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d5)
	Kernel driver in use: pcieport
00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 [8086:8c16] (rev d5)
	Kernel driver in use: pcieport
00:1f.0 ISA bridge [0601]: Intel Corporation HM87 Express LPC Controller [8086:8c4b] (rev 05)
	Subsystem: Intel Corporation HM87 Express LPC Controller [8086:7270]
	Kernel driver in use: lpc_ich
	Kernel modules: lpc_ich
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
	Subsystem: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:7270]
	Kernel driver in use: i801_smbus
	Kernel modules: i2c_i801
00:1f.6 Signal processing controller [1180]: Intel Corporation 8 Series Chipset Family Thermal Management Controller [8086:8c24] (rev 05)
	Subsystem: Intel Corporation 8 Series Chipset Family Thermal Management Controller [8086:7270]
	Kernel driver in use: intel_pch_thermal
	Kernel modules: intel_pch_thermal
01:00.0 SATA controller [0106]: Samsung Electronics Co Ltd Device [144d:a801] (rev 01)
	Subsystem: Samsung Electronics Co Ltd Device [144d:a801]
	Kernel driver in use: ahci
	Kernel modules: ahci
03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM43602 802.11ac Wireless LAN SoC [14e4:43ba] (rev 01)
	Subsystem: Apple Inc. BCM43602 802.11ac Wireless LAN SoC [106b:0133]
	Kernel driver in use: brcmfmac
	Kernel modules: brcmfmac
04:00.0 Multimedia controller [0480]: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera [14e4:1570]
	Subsystem: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera [14e4:1570]
	Kernel driver in use: facetimehd
	Kernel modules: facetimehd
05:00.0 PCI bridge [0604]: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013] [8086:156d]
	Kernel driver in use: pcieport
06:00.0 PCI bridge [0604]: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013] [8086:156d]
	Kernel driver in use: pcieport
06:03.0 PCI bridge [0604]: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013] [8086:156d]
	Kernel driver in use: pcieport
06:04.0 PCI bridge [0604]: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013] [8086:156d]
	Kernel driver in use: pcieport
06:05.0 PCI bridge [0604]: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013] [8086:156d]
	Kernel driver in use: pcieport
06:06.0 PCI bridge [0604]: Intel Corporation DSL5520 Thunderbolt 2 Bridge [Falcon Ridge 4C 2013] [8086:156d]
	Kernel driver in use: pcieport
07:00.0 System peripheral [0880]: Intel Corporation DSL5520 Thunderbolt 2 NHI [Falcon Ridge 4C 2013] [8086:156c]
	Kernel driver in use: thunderbolt
	Kernel modules: thunderbolt

```

ledisthebest, 2022-06-07:
A new update was issued few days ago, have you try to re-clone this repo?

mirfan899, 2022-06-08:
Yes, I have tried the latest clone.
[image]

Still getting green screen on cheese camera, and black on skype.
[image]