patjak/facetimehd #268: Cheese not working on my Macbook Air 7,2

JuicyJammy, 2022-06-13:
I'm using Ubuntu 22.04, and when I installed this module, the camera worked.
But Cheese would freeze, and it'll display things that's not moving.
When I run cheese from command line, it says something about a "timestrap problem".
Can anyone help?

zoof, 2022-06-21:
I've observed the same with both Cheese and Zoom.  Was not a problem with Debian Bullseye.  The following is the output from running Cheese from the command line.
```
$ cheese

(cheese:5986): Gdk-CRITICAL **: 08:28:51.760: gdk_wayland_window_set_dbus_properties_libgtk_only: assertion 'GDK_IS_WAYLAND_WINDOW (window)' failed

(cheese:5986): cheese-WARNING **: 08:28:56.487: A lot of buffers are being dropped.: ../libs/gst/base/gstbasesink.c(3143): gst_base_sink_is_too_late (): /GstCameraBin:camerabin/GstViewfinderBin:vf-bin/ClutterGstVideoSink:cluttergstvideosink0:
There may be a timestamping problem, or this computer is too slow.

(cheese:5986): Clutter-CRITICAL **: 08:29:09.107: Unable to create dummy onscreen: No foreign surface, and wl_shell unsupported by the compositor
```

pasan93, 2022-06-28:
@JuicyJammy Were you able to fix it? I couldn't even install the driver on Pop!_OS 22.04. 😞

pasan93, 2022-07-06:
> > @JuicyJammy Were you able to fix it? I couldn't even install the driver on Pop!_OS 22.04. disappointed
> 
> I am also on Pop!_OS 22.04 and had no problems building the driver on the current 5.17 kernel. But I get the same image freeze that @JuicyJammy is getting.

I tried the driver even by distro hopping from Pop!_OS to Ubuntu and Fedora.
Still no luck.

satmandu, 2022-07-21:
Looks like this is a cheese issue upstream: https://gitlab.gnome.org/GNOME/cheese/-/issues/126  (That matches some of the issues, but I don't have the choppy video. Cheese just freezes for me.)

This may work fine for you: `GST_V4L2_USE_LIBV4L2=1 gst-launch-1.0 v4l2src ! xvimagesink` (It works fine for me.)

zoof, 2022-07-27:
@satmandu as I noted, video on Zoom doesn't work properly either (with Pop!_OS) so it's not obvious that it is purely a Cheese issue.  I also tried to get video to work with Arch, using the LTS kernel (5.15?) and Debian testing (Bookworm) but no luck (have not tried Zoom with these options).  I am back to Debian stable where I know everything works.

zoof, 2022-07-31:
I installed the backported kernel and headers for Bullseye (v 5.18) and Cheese works so as suggested by @satmandu, it seems like it is a problem with some dependency upstream of both Cheese and Zoom.

jmsunseri, 2023-03-18:
FWIW I'm on Fedora with kernel 6.1.18-200.fc37.x86_64.  Zoom works perfectly but Cheese still has the same issue

amunizp, 2023-10-24:
hoping to add a bit more info here with Apple Inc. MacBookAir7,2 (apple description: MacBook Air (13-inch, 2017)) using ubuntu Ubuntu 22.04.3 LTS

I installed (see notes below on how I installed)  and confirm the following:

**Cheese** image shows but freezes and then crashes. 
**Kamoso** Seems to do picture with wrong colours (i get purple and green with vertical lines) and video does not work. 
**guvcview** Seems to work fine but commandline outputs loads of warnings that it seems to ignore. 
**googlemeet via Chrome Version 118.0.5993.88 (Official Build) (64-bit) ** Seems to work fine. 
**googlemeet via firefox 118.0.2 (64-bit) snap for ubuntu** Seems to work fine. 
** zoom on browsers (as above)** seems to work fine. 

```
$ cheese

(cheese:2292): Gdk-WARNING **: 15:05:01.822: Native Windows taller than 65535 pixels are not supported

(cheese:2292): cheese-WARNING **: 15:05:06.497: A lot of buffers are being dropped.: ../libs/gst/base/gstbasesink.c(3143): gst_base_sink_is_too_late (): /GstCameraBin:camerabin/GstViewfinderBin:vf-bin/ClutterGstVideoSink:cluttergstvideosink0:
There may be a timestamping problem, or this computer is too slow.

(cheese:2292): cheese-WARNING **: 15:05:24.986: Output state was not configured: ../gst-libs/gst/video/gstvideoencoder.c(2281): gst_video_encoder_can_push_unlocked (): /GstCameraBin:camerabin/GstEncodeBin:image-encodebin/GstJpegEnc:jpegenc0

(cheese:2292): Clutter-CRITICAL **: 15:05:30.142: Unable to create dummy onscreen: No foreign surface, and wl_shell unsupported by the compositor
$ kamoso 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
kf.kirigami: Warning: Theme implementations should use Kirigami.BasicThemeDefinition for its root item
kf.kirigami: Failed to find a Kirigami platform plugin
Creating QUrl("file:///home/andres/Pictures")
skipping message... imagebin-filesink
kf.kio.widgets: Failed to check which JobView API is supported "The name org.kde.kuiserver was not provided by any .service files"
Killed
$ guvcview 
V4L2_CORE: Unable to find parent usb device.GUVCVIEW: version 2.0.7
GUVCVIEW: couldn't open /home/andres/.config/guvcview2/video0 for read: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Inappropriate ioctl for device
ALSA lib pcm_dsnoop.c:601:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2664:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
control[0]:(unknown - 0x6) 0x980001 'User Controls'
[... 2 lines cut ...]
ENCODER: add stream 0 to stream list
ENCODER: add stream 1 to stream list
ENCODER: (matroska) add seekhead entry 0 (max 10)
ENCODER: (matroska) add seekhead entry 1 (max 10)
[mp2 @ 0x7fd2c41cee40] Ignoring attempt to flush encoder that doesn't support it
AUDIO: (portaudio) Aborting audio stream
AUDIO: Closing audio stream...
[mpeg4 @ 0x7fd2c4001800] Ignoring attempt to flush encoder that doesn't support it
ENCODER: (matroska) closing context
ENCODER: (matroska) closing cluster
ENCODER: (matroska)writing cues
ENCODER: (matroska)add seekhead
ENCODER: (matroska) add seekhead entry 2 (max 10)
ENCODER: (matroska)write seekhead
ENCODER: (matroska) end duration = 3872 (3872.000000) 
[mp2 @ 0x7fd2c41cee40] Ignoring attempt to flush encoder that doesn't support it
```

These are the steps I followed to install on my machine. 

`git clone https://github.com/patjak/bcwc_pcie.git`

Then created the script for ubuntu as described here: https://github.com/patjak/facetimehd/wiki/Installation#ubuntu-installation-script

I did not check if it was needed but also did the suspend steps here: https://github.com/patjak/facetimehd/wiki/Installation#for-ubuntu-1504-and-higher-should-work-on-some-other-distros-too