patjak/facetimehd #275: attempting to install on ubuntu 22.04 early 2015 macbook air

geoihs, 2022-09-10:
UbuntuAir:~/bcwc_pcie$ sudo make install
make -C /lib/modules/5.15.0-47-generic/build M=/home/geo/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-47-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.15.0-47-generic/extra/facetimehd.ko
  SIGN    /lib/modules/5.15.0-47-generic/extra/facetimehd.ko
At main.c:160:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: ../crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: ../crypto/bio/bss_file.c:75
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.15.0-47-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'

  I've been away from linux for a LONG time. Any help appreciated?

larsblumberg, 2022-12-19:
Here's what worked for me, I'm successfully using `facethimehd` on an early 2015 macbook air model 7,1 on Pop_OS! 22.04. Since Pop_OS! is built on Ubuntu 22.04, this should work for you, too:

```
modprobe -D facetimehd \
  && { echo "facetimehd driver already installed"; exit 0; }
[[ ! -d "facetimehd" ]] \
  && git clone https://github.com/patjak/facetimehd.git
[[ ! -d "facetimehd-firmware" ]] \
  && git clone https://github.com/patjak/facetimehd-firmware.git
# Compile driver
cd facetimehd-firmware
make
sudo make install
# Install driver
cd ../facetimehd
# Create signing_key.pem, required by driver installation
# https://superuser.com/a/1322832
certs_directory=$(find /usr/src/*-generic/certs | head -n 1)
[[ -z "certs_directory" ]] && { echo "Certs directory not found"; exit 1; }
if [[ ! -f "$certs_directory/signing_key.pem" ]]; then
  tee signing_key.config << EOF
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts

[ req_distinguished_name ]
CN = Modules

[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid
EOF
  openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config signing_key.config -outform DER -out signing_key.x509 -keyout signing_key.pem
  sudo mv signing_key.pem signing_key.x509 $certs_directory
fi
# Install driver
make
sudo make install
sudo depmod
sudo modprobe facetimehd
rg -q facetimehd /etc/modules \
  || echo facetimehd | sudo tee -a /etc/modules
# Clean up
cd ..
rm -rf facetimehd facetimehd-firmware
```

Sources:
* https://askubuntu.com/a/1380378
* https://superuser.com/a/1322832

oliv3r, 2022-12-26:
While I'm not using your install script, the module has stopped working for me, I think since kernel version 6.0. I suppose Ubuntu is using still an older kernel?

larsblumberg, 2022-12-26:
I've read somewhere that the driver needs to be rebuilt with every kernel upgrade. My current kernel version is

```
❯ uname -r
6.0.6-76060006-generic
```

and the facetime module works fine.

Was your kernel recently upgraded? Did you then reinstall the module?

oliv3r, 2022-12-27:
I'm on arch, and indeed, I have to rebuild it every upgrade :) it's now 'muscle memory' so yes, I always reinstall it. (Also not rebuilding it means i can't find the module anyway :p)

andrewpking, 2025-01-30:
> Here's what worked for me, I'm successfully using `facethimehd` on an early 2015 macbook air model 7,1 on Pop_OS! 22.04. Since Pop_OS! is built on Ubuntu 22.04, this should work for you, too:
>
> ``` [...]

This script should be in the official docs for Ubuntu installation. Much better than copy pasting one command at a time. Also, kudos to you for automating this process.

larsblumberg, 2025-01-31:
> Much better than copy pasting one command at a time.

You should be able to put all the commands into a single shell script file, then make it executable and run it. So that you avoid running command by command only.