patjak/facetimehd #278: No longer working on kernel version 6.0 or 6.1

oliv3r, 2022-12-26:
Compilation doesn't produce any errors or warnings, however cheese reports the following:

```
(cheese:5211): cheese-WARNING **: 11:48:15.090: Internal data stream error.: ../gstreamer/subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin35/GstPipeWireSrc:pipewiresrc1:
streaming stopped, reason not-negotiated (-4)
```
Which doesn't really help either.

Probing the module yields:
```
[ 1392.221914] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 1392.222063] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 1392.226919] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 1392.226945] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 1392.236957] facetimehd 0000:02:00.0: PLL reset finished
[ 1392.236959] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 1392.236971] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[ 1392.246977] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[ 1392.267013] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 1392.267027] facetimehd 0000:02:00.0: STRAP valid
[ 1392.267028] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 1392.267035] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 1392.267046] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 3 us
[ 1392.267056] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 1392.267057] facetimehd 0000:02:00.0: Using step size 145
[ 1392.267061] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x10119
[ 1392.267066] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 1392.287587] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 1392.287601] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 1392.287795] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 1392.402443] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[ 1392.557320] facetimehd 0000:02:00.0: ISP woke up after 0ms
[ 1392.557327] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[ 1392.557331] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[ 1392.567345] facetimehd 0000:02:00.0: ISP second int after 0ms
[ 1392.567347] facetimehd 0000:02:00.0: Channel description table at 00800000
[ 1392.577594] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[ 1392.577596] facetimehd 0000:02:00.0: Enabling interrupts
[ 1392.577763] FWMSG: 
[ 1392.577795] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[ 1392.577805] FWMSG: 
[ 1392.577819] FWMSG:   Restart count : 0
[ 1392.577896] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[ 1392.577914] FWMSG:   System Clock : 200000000 Hz
[ 1392.577931] FWMSG:   Processor mode : SUPERVISOR
[ 1392.577951] FWMSG:   Cache architecture type : SEPARATE
[ 1392.577967] FWMSG:   Cache type : WRITEBACK
[ 1392.577985] FWMSG:   Data Cache Line : 32 [0x20]
[ 1392.578002] FWMSG:   Boot arguments entries : 2
[ 1392.578065] FWMSG:     0000: 0x00000000 0x00000000
[ 1392.578091] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[ 1392.578110] FWMSG:   Main memory :   base : 0x00000000
[ 1392.578136] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[ 1392.578161] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[ 1392.578251] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[ 1392.578276] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[ 1392.578310] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[ 1392.578329] FWMSG:   Shared memory : base : 0x00800000
[ 1392.578356] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[ 1392.578480] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[ 1392.578510] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[ 1392.578531] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[ 1392.578552] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[ 1392.578619] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[ 1392.578637] FWMSG:   Application : adc [release]
[ 1392.578657] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[ 1392.578674] FWMSG:   Release : S2ISP-01.43.00
[ 1392.578689] FWMSG:   H4ISPAPPLE : 11536
[ 1392.578703] FWMSG:   H4ISPCD : 4081
[ 1392.578716] FWMSG:   ffw : 4143
[ 1392.578740] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[ 1392.578749] FWMSG: 
[ 1392.578800] FWMSG: -------------------------------------------------------------------------------
[ 1392.578809] FWMSG: 
[ 1392.652039] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[ 1392.652205] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[ 1392.652393] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[ 1392.654244] FWMSG: [ISP] CH = 0 CMD = 0x0111 [CISP_CMD_CH_SET_FILE_LOAD]
```
which doesn't seem troublesome ...

The driver was working quite well before ...

larsblumberg, 2023-01-18:
```
> uname -r
6.0.6-76060006-generic

oliv3r, 2023-01-18:
Seems to be working again. I don't know why it wasn't working before ... Maybe an arch bug :) Sorry for the noise, but hope to see it upstreamed soon :p

closed 2023-01-18 by oliv3r