patjak/facetimehd #283: pci_set_dma_mask and pci_set_consistent_dma_mask depreciation

pswindells77, 2023-04-09:
Hi, linked to the issue raised previously, issue #282.
I noticed that pci_set_dma_mask and pci_set_consistent_dma_mask have recently been depreciated.
Can anybody help with a workaround?  I am pretty confident this is the cause of issue #282.
Thanks.

pswindells77, 2023-05-04:
Camera now working again.  I replaced call on line 399 with 
// pci_set_consistent_dma_mask(pdev, DMA_BIT_MASK(dev_priv->dma_mask));
 dma_set_coherent_mask(&pdev->dev, DMA_BIT_MASK(dev_priv->dma_mask));

closed 2023-05-04 by pswindells77

patjak, 2023-05-09:
This should have been fixed in 75a2a5800aa46cbd679847b4d0863a5e3cef3b9e
Perhaps you have a distro kernel with the dma_set_coherent_mask() change backported but still reports as a pre v5.18 kernel.