patjak/facetimehd #285: facetimehd not working in Debian Bookworm

rgrove-rsu20, 2023-05-02:
facetimehd doesn't seem to be working in Bookworm. I think this is how I've installed this in the past (using DKMS in Bullseye):
apt -y install dkms linux-headers-amd64 git kmod libssl-dev checkinstall
wget https://github.com/patjak/facetimehd/archive/refs/tags/0.5.18.tar.gz
tar xf 0.5.18.tar.gz -C /usr/src/
dkms add -m facetimehd -v 0.5.18
dkms build -m facetimehd -v 0.5.18
dkms install -m facetimehd -v 0.5.18
echo "facetimehd" >> /etc/modules

rgrove-rsu20, 2023-05-02:
[  +0.005293] facetimehd: loading out-of-tree module taints kernel.
[  +0.000067] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[  +0.001396] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  +0.000161] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  +0.005884] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  +0.000031] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  +0.009934] facetimehd 0000:02:00.0: PLL reset finished
[  +0.000006] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  +0.000014] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  +0.009928] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  +0.019879] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  +0.000016] facetimehd 0000:02:00.0: STRAP valid
[  +0.000002] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  +0.000007] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  +0.000010] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  +0.000010] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  +0.000001] facetimehd 0000:02:00.0: Using step size 150
[  +0.000005] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[  +0.000005] facetimehd 0000:02:00.0: Virtual VTT enabled
[  +0.020354] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  +0.000016] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  +0.000197] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  +0.000481] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/firmware.bin (-2)
[  +0.000003] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[  +0.000012] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/firmware.bin (-2)
[  +0.000002] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[  +0.000122] facetimehd: probe of 0000:02:00.0 failed with error -2

rgrove-rsu20, 2023-05-03:
Ok, I was running:
https://raw.githubusercontent.com/patjak/facetimehd-firmware/master/facetimehd-firmware-install.sh

However, after running:
git clone https://github.com/patjak/facetimehd-firmware.git
cd facetimehd-firmware
make
make install

It still doesn't work for me.

rgrove-rsu20, 2023-05-03:
[    6.900079] facetimehd: loading out-of-tree module taints kernel.
[    6.900152] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[    6.901686] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[    6.901848] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[    6.907202] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[    6.907233] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[    6.917166] facetimehd 0000:02:00.0: PLL reset finished
[    6.917170] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[    6.917183] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[    6.927108] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[    6.946981] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[    6.946995] facetimehd 0000:02:00.0: STRAP valid
[    6.946996] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[    6.947003] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[    6.947013] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[    6.947023] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[    6.947024] facetimehd 0000:02:00.0: Using step size 153
[    6.947029] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[    6.947034] facetimehd 0000:02:00.0: Virtual VTT enabled
[    6.967387] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[    6.967401] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[    6.967595] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[    6.973071] facetimehd 0000:02:00.0: firmware: direct-loading firmware facetimehd/firmware.bin
[    7.087271] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[    7.244628] facetimehd 0000:02:00.0: ISP woke up after 0ms
[    7.244638] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[    7.270823] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[    7.280770] facetimehd 0000:02:00.0: ISP second int after 0ms
[    7.280788] facetimehd 0000:02:00.0: Channel description table at 00800000
[    7.292694] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[    7.292701] facetimehd 0000:02:00.0: Enabling interrupts
[    7.293498] FWMSG: 
[    7.293535] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[    7.293547] FWMSG: 
[    7.293562] FWMSG:   Restart count : 0
[    7.293592] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[    7.293610] FWMSG:   System Clock : 200000000 Hz
[    7.293629] FWMSG:   Processor mode : SUPERVISOR
[    7.293651] FWMSG:   Cache architecture type : SEPARATE
[    7.293668] FWMSG:   Cache type : WRITEBACK
[    7.293687] FWMSG:   Data Cache Line : 32 [0x20]
[    7.293706] FWMSG:   Boot arguments entries : 2
[    7.293724] FWMSG:     0000: 0x00000000 0x00000000
[    7.293751] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[    7.293771] FWMSG:   Main memory :   base : 0x00000000
[    7.293799] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[    7.293826] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[    7.293853] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[    7.293878] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[    7.293906] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[    7.293926] FWMSG:   Shared memory : base : 0x00800000
[    7.293954] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[    7.293985] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[    7.294015] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[    7.294038] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[    7.294061] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[    7.294089] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[    7.294108] FWMSG:   Application : adc [release]
[    7.294128] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[    7.294147] FWMSG:   Release : S2ISP-01.43.00
[    7.294163] FWMSG:   H4ISPAPPLE : 11536
[    7.294178] FWMSG:   H4ISPCD : 4081
[    7.294192] FWMSG:   ffw : 4143
[    7.294217] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[    7.294227] FWMSG: 
[    7.294260] FWMSG: -------------------------------------------------------------------------------
[    7.294270] FWMSG: 
[    7.366954] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[    7.367073] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[    7.367301] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[    7.367407] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/1871_01XX.dat (-2)
[    7.367427] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/1871_01XX.dat (-2)
[    7.367431] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

patjak, 2023-05-03:
The firmware is now loading. The sensor files are not (errors at the bottom) but they are not required. Are you still not getting a /dev/video0 device?

rgrove-rsu20, 2023-05-03:
yes, a /dev/video0 device is created, however nothing seems to be able to use it. The launch of Cheese hangs for a bit and comes back with 'There was an error playing video from the webcam'. I have both MacBookAir7,2 and MacBookAir7,1 models and both exhibit the same behavior.

rgrove-rsu20, 2023-05-03:
journalctl contains:

pipewire[3451]: spa.v4l2: '/dev/video0' can't allocate enough buffers (4 < 16)
pipewire[3451]: pw.port: 0x556b7b5fa760: negotiate buffers on node: -95 (Operation not supported)
pipewire[3451]: pw.port: 0x556b7b5fa760: state ready -> error (can't negotiate buffers on port)
pipewire[3451]: pw.link: (43.0 -> 56.0) allocating -> error (can't negotiate buffers on port)