patjak/facetimehd #290: macbook 2015

MaxnoSPB, 2023-10-14:
My config is https://linux-hardware.org/?probe=1a8e527488

I install the facetime driver as you discribe 
But its dont work 

 mplayer tv://
MPlayer 1.4 (Debian), built with gcc-11 (C) 2000-2019 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!
v4l2: ioctl queue buffer failed: No buffer space available
v4l2: ioctl set mute failed: Invalid argument
v4l2: 0 frames successfully processed, 0 frames dropped.

Exiting... (End of file)

Discord and google meets dont work too

TotalTaxAmount, 2024-10-14:
Did you ever resolve this issue? I have the same thing

MaxnoSPB, 2024-10-15:
Yeah, it turned out to be pretty simple. I broke that old mac and bought a new one. Everything works now.