patjak/facetimehd #293: "make" command while in bcwc_pcie directory does not work

mavelwort, 2024-03-10:
I'm trying to install the driver and I cannot get the make command to work when I want to build the kernel module in bcwc_pcie.
Here's the error code I keep getting:
```
make -C /lib/modules/6.5.0-15-generic/build M=/home/pancake/bcwc_pcie modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-15-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           
  CC [M]  /home/pancake/bcwc_pcie/fthd_ddr.o
/bin/sh: 1: gcc-12: not found
make[3]: *** [scripts/Makefile.build:251: /home/pancake/bcwc_pcie/fthd_ddr.o] Error 127
make[2]: *** [/usr/src/linux-headers-6.5.0-15-generic/Makefile:2037: /home/pancake/bcwc_pcie] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-15-generic'
make: *** [Makefile:12: all] Error 2
```

I'm rather new to Linux and therefore not too familiar with the terminal, so I don't understand what it means telling me that the compiler differs. Any help would be _greatly_ appreciated.

jonwilts, 2024-03-11:
Looks like this is the cause: [https://lkml.iu.edu/hypermail/linux/kernel/2312.0/06197.html](https://lkml.iu.edu/hypermail/linux/kernel/2312.0/06197.html)

I was able to build the module on Arch by doing the following:

    cd /var/lib/dkms/bcwc-pcie/0.6.8.r2.g2b287d4/build
    sudo vi fthd_v4l2.c

Then change the line: 

    q->min_buffers_needed = 1;

to

    q->min_queued_buffers = 1;

and build with 

    sudo KERNELRELEASE=6.8.0-zen1-1-zen make
    sudo mkinitcpio -P

mulderje, 2024-03-19:
Can confirm that updating `s/min_buffers_needed/min_queued_buffers/`  worked for me on the`6.9.0-0.rc0.20240314git480e035fc4c7.5.fc41.x86_64` (Fedora Rawhide) kernel. Thank you for the pointer, @jonwilts.

mavelwort, 2024-03-19:
> Looks like this is the cause: https://lkml.iu.edu/hypermail/linux/kernel/2312.0/06197.html
> 
> I was able to build the module on Arch by doing the following:
> 
> ```
> cd /var/lib/dkms/bcwc-pcie/0.6.8.r2.g2b287d4/build
> sudo vi fthd_v4l2.c
> ```

The directory you want me to enter does not exist on my laptop. Is this due to you using another distro or...?

mulderje, 2024-03-30:
FYI this was fixed in https://github.com/patjak/facetimehd/pull/292 for me. Should just need to update your source.

mavelwort, 2024-04-04:
Sorry if this seems like a dumb question but how do i do that? Update my source, I mean. I see that there was an update about 16-17 hours ago so I deleted the already existing folders of "facetimehd-firmware" and "bcwc_pcie", then went through the installation steps again. Once I got to using the "make" command in the bcwc_pcie directory, I encountered the same error.

I also finally understood what you meant in this step and though the first line of code doesn't work for me, I found the file and manually changed the line.

> ```
> cd /var/lib/dkms/bcwc-pcie/0.6.8.r2.g2b287d4/build
> sudo vi fthd_v4l2.c
> ```
> 
> Then change the line:
> 
> ```
> q->min_buffers_needed = 1;
> ```
> 
> to
> 
> ```
> q->min_queued_buffers = 1;
> ```

The only issue is that this didn't work for me. I could not build with it.
> and build with
> ```
> sudo KERNELRELEASE=6.8.0-zen1-1-zen make
> sudo mkinitcpio -P
> ```

Thus, I can't still can't use the make command in the bcwc_pcie directory.