patjak/facetimehd #295: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

yaguphire, 2024-05-12:
no issues when compiling facetimehd.  camera does work but noticing that colors are off showing green and pink.

when loading the facetimehd module

May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: Setting 64bit DMA mask
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: S2 PCIe link init succeeded
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: PLL reset finished
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
May 12 16:29:37 tux kernel: facetimehd 0000:02:00.0: S2 PLL is in bypass mode
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: STRAP valid
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Using step size 149
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Virtual VTT enabled
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: ISP woke up after 0ms
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: ISP second int after 0ms
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Channel description table at 00800000
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Enabling interrupts
May 12 16:29:38 tux kernel: FWMSG: 
May 12 16:29:38 tux kernel: FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
May 12 16:29:38 tux kernel: FWMSG: 
May 12 16:29:38 tux kernel: FWMSG:   Restart count : 0
May 12 16:29:38 tux kernel: FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
May 12 16:29:38 tux kernel: FWMSG:   System Clock : 200000000 Hz
May 12 16:29:38 tux kernel: FWMSG:   Processor mode : SUPERVISOR
May 12 16:29:38 tux kernel: FWMSG:   Cache architecture type : SEPARATE
May 12 16:29:38 tux kernel: FWMSG:   Cache type : WRITEBACK
May 12 16:29:38 tux kernel: FWMSG:   Data Cache Line : 32 [0x20]
May 12 16:29:38 tux kernel: FWMSG:   Boot arguments entries : 2
May 12 16:29:38 tux kernel: FWMSG:     0000: 0x00000000 0x00000000
May 12 16:29:38 tux kernel: FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
May 12 16:29:38 tux kernel: FWMSG:   Main memory :   base : 0x00000000
May 12 16:29:38 tux kernel: FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
May 12 16:29:38 tux kernel: FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
May 12 16:29:38 tux kernel: FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
May 12 16:29:38 tux kernel: FWMSG:   Shared window : base : 0x00800000 [static wiring]
May 12 16:29:38 tux kernel: FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
May 12 16:29:38 tux kernel: FWMSG:   Shared memory : base : 0x00800000
May 12 16:29:38 tux kernel: FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
May 12 16:29:38 tux kernel: FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
May 12 16:29:38 tux kernel: FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
May 12 16:29:38 tux kernel: FWMSG:   Heap free space : 10002626 [0x0098a0c2]
May 12 16:29:38 tux kernel: FWMSG:   Heap allocated space : 94656 [0x000171c0]
May 12 16:29:38 tux kernel: FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
May 12 16:29:38 tux kernel: FWMSG:   Application : adc [release]
May 12 16:29:38 tux kernel: FWMSG:   Linked on : Jul 25 2015 - 08:48:55
May 12 16:29:38 tux kernel: FWMSG:   Release : S2ISP-01.43.00
May 12 16:29:38 tux kernel: FWMSG:   H4ISPAPPLE : 11536
May 12 16:29:38 tux kernel: FWMSG:   H4ISPCD : 4081
May 12 16:29:38 tux kernel: FWMSG:   ffw : 4143
May 12 16:29:38 tux kernel: FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
May 12 16:29:38 tux kernel: FWMSG: 
May 12 16:29:38 tux kernel: FWMSG: -------------------------------------------------------------------------------
May 12 16:29:38 tux kernel: FWMSG: 
May 12 16:29:38 tux kernel: FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
May 12 16:29:38 tux kernel: FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
May 12 16:29:38 tux kernel: FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
May 12 16:29:38 tux kernel: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

yaguphire, 2024-05-12:
i forgot to mention on the latest kernel.  6.8.9

yaguphire, 2024-05-12:
ran this after reading other issues people were having.  here is the output modprobe facetimehd dyndbg==pmf

May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
May 12 16:51:27 tux kernel: facetimehd:fthd_pci_reserve_mem: Allocated S2 regs (BAR 0). 65536 bytes at 0x00000000f5f42597
May 12 16:51:27 tux kernel: facetimehd:fthd_pci_reserve_mem: Allocated S2 mem (BAR 2). 268435456 bytes at 0x000000004d551387
May 12 16:51:27 tux kernel: facetimehd:fthd_pci_reserve_mem: Allocated ISP regs (BAR 4). 1048576 bytes at 0x000000006341670d
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Setting 64bit DMA mask
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: S2 PCIe link init succeeded
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: PLL reset finished
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: S2 PLL is in bypass mode
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: STRAP valid
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Using step size 149
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Virtual VTT enabled
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: ISP woke up after 0ms
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: ISP second int after 0ms
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Channel description table at 00800000
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 0: TERMINAL, type 2, source 0, size 512, offset 800700
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 1: IO, type 0, source 1, size 8, offset 808700
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 2: DEBUG, type 0, source 1, size 8, offset 808900
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 3: BUF_H2T, type 0, source 2, size 64, offset 808b00
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 4: BUF_T2H, type 1, source 3, size 64, offset 809b00
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 5: SHAREDMALLOC, type 1, source 3, size 8, offset 80ab00
May 12 16:51:27 tux kernel: facetimehd:isp_fill_channel_info: Channel 6: IO_T2H, type 1, source 2, size 8, offset 80ad00
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_init: clearing ringbuf IO at 00808700 (size 8)
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_init: clearing ringbuf DEBUG at 00808900 (size 8)
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_init: clearing ringbuf BUF_H2T at 00808b00 (size 64)
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Enabling interrupts
May 12 16:51:27 tux kernel: facetimehd:fthd_isp_cmd_start: sending start cmd to firmware
May 12 16:51:27 tux kernel: facetimehd:fthd_isp_cmd: sending cmd 0 to firmware
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_send: send 00b0c000
May 12 16:51:27 tux kernel: facetimehd:fthd_handle_irq: channel SHAREDMALLOC: message available, address 00000000
May 12 16:51:27 tux kernel: facetimehd:sharedmalloc_handler: Firmware allocated 65536 bytes at 00b0d000 (tag \x00LOG)
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_send: send 00b0d040
May 12 16:51:27 tux kernel: facetimehd:fthd_handle_irq: channel TERMINAL: message available, address 00b0d042
May 12 16:51:27 tux kernel: FWMSG: 
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_send: send 00000000
May 12 16:51:27 tux kernel: facetimehd:fthd_handle_irq: channel TERMINAL: message available, address 00b0d0c2
May 12 16:51:27 tux kernel: FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_send: send 00000000
May 12 16:51:27 tux kernel: facetimehd:fthd_handle_irq: channel TERMINAL: message available, address 00b0d142
May 12 16:51:27 tux kernel: FWMSG: 
May 12 16:51:27 tux kernel: facetimehd:fthd_channel_ringbuf_send: send 00000000
May 12 16:51:27 tux kernel: facetimehd:fthd_handle_irq: channel TERMINAL: message available, address 00b0d1c2
[... cut ...]
May 12 16:51:27 tux kernel: facetimehd:fthd_isp_cmd_channel_info: sensor serial number: EFF40100400D0300
May 12 16:51:27 tux kernel: facetimehd:fthd_isp_cmd_set_loadfile: set loadfile
May 12 16:51:27 tux kernel: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
May 12 16:51:27 tux kernel: facetimehd:fthd_v4l2_ioctl_enum_frameintervals: fthd_v4l2_ioctl_enum_frameintervals
May 12 16:51:27 tux kernel: facetimehd:fthd_v4l2_ioctl_enum_frameintervals: fthd_v4l2_ioctl_enum_frameintervals

AlexG-78, 2024-06-18:
Same happens to me as well. I am on Ubuntu 24.04 with Kernel 6.8.0-35-generic #35-Ubuntu SMP PREEMPT_DYNAMIC

[    2.226971] facetimehd 0000:05:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
 I am on a Macbook Pro 15" Mid 2015

AlexG-78, 2024-06-22:
> Same happens to me as well. I am on Ubuntu 24.04 with Kernel 6.8.0-35-generic Acually the camra is not working. Neither Cheese nor Snapshot are recognizing the camera. "No device found"
> 
> [ 2.226971] facetimehd 0000:05:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2 I am on a Macbook Pro 15" Mid 2015

Got it to work. Finally https://github.com/patjak/facetimehd/issues/163#issuecomment-428712954 resolved my issue. I needed to be part of the video group

satmandu, 2025-03-09:
I'm also getting the `facetimehd 0000:04:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2` message. 

I'm only able to see one image though. Video does not work.  (I am using the current version of the driver with kernel 6.14.0-rc5).

I have extracted the firmware, and if I do `rmmod facetimehd && modprobe facetimehd` I don't get the firmware load message a second time.

patjak, 2026-03-20:
This error is harmless. See known issues in the wiki. If you really want to fix it see: https://github.com/patjak/facetimehd/wiki/Extracting-the-sensor-calibration-files