patjak/facetimehd #297: Build failure Kernel 6.8.0

tr33, 2024-07-23:
## description

Builing the module fails with Linux kernel 6.7 and 6.8 fails.
It seems like that the media/videobuf-dma-sg.h file it's complaining about is no longer in kernel 6.7/6.8.
(see https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/common/videobuf2/videobuf2-dma-sg.c?h=v6.7)

## Steps to reproduce

Steps to reproduce the behavior:

    build facetimehd

## Additional context

Linux kernel 6.8.0-38-generic Ubuntu

## Build log
```
ProblemType: Package
DKMSBuildLog:
 DKMS make.log for facetimehd-0.1 for kernel 6.8.0-38-generic (x86_64)
 So 21. Jul 14:41:27 CEST 2024
 make: Verzeichnis „/usr/src/linux-headers-6.8.0-38-generic“ wird betreten
 warning: the compiler differs from the one used to build the kernel
   The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
   You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
   CC [M]  /var/lib/dkms/facetimehd/0.1/build/fthd_ddr.o
   CC [M]  /var/lib/dkms/facetimehd/0.1/build/fthd_hw.o
   CC [M]  /var/lib/dkms/facetimehd/0.1/build/fthd_drv.o
   CC [M]  /var/lib/dkms/facetimehd/0.1/build/fthd_ringbuf.o
 In file included from /var/lib/dkms/facetimehd/0.1/build/fthd_drv.h:24,
                  from /var/lib/dkms/facetimehd/0.1/build/fthd_ddr.c:21:
 /var/lib/dkms/facetimehd/0.1/build/fthd_v4l2.h:17:10: fatal error: media/videobuf-dma-sg.h: Datei oder Verzeichnis nicht gefunden
    17 | #include <media/videobuf-dma-sg.h>
       |          ^~~~~~~~~~~~~~~~~~~~~~~~~
 compilation terminated.
 In file included from /var/lib/dkms/facetimehd/0.1/build/fthd_drv.h:24,
                  from /var/lib/dkms/facetimehd/0.1/build/fthd_ringbuf.c:15:
 /var/lib/dkms/facetimehd/0.1/build/fthd_v4l2.h:17:10: fatal error: media/videobuf-dma-sg.h: Datei oder Verzeichnis nicht gefunden
    17 | #include <media/videobuf-dma-sg.h>
       |          ^~~~~~~~~~~~~~~~~~~~~~~~~
 compilation terminated.
 In file included from /var/lib/dkms/facetimehd/0.1/build/fthd_drv.h:24,
                  from /var/lib/dkms/facetimehd/0.1/build/fthd_hw.c:11:
 /var/lib/dkms/facetimehd/0.1/build/fthd_v4l2.h:17:10: fatal error: media/videobuf-dma-sg.h: Datei oder Verzeichnis nicht gefunden
    17 | #include <media/videobuf-dma-sg.h>
       |          ^~~~~~~~~~~~~~~~~~~~~~~~~
 compilation terminated.
 make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/facetimehd/0.1/build/fthd_ddr.o] Fehler 1
 make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
 In file included from /var/lib/dkms/facetimehd/0.1/build/fthd_drv.h:24,
                  from /var/lib/dkms/facetimehd/0.1/build/fthd_drv.c:25:
 /var/lib/dkms/facetimehd/0.1/build/fthd_v4l2.h:17:10: fatal error: media/videobuf-dma-sg.h: Datei oder Verzeichnis nicht gefunden
    17 | #include <media/videobuf-dma-sg.h>
       |          ^~~~~~~~~~~~~~~~~~~~~~~~~
 compilation terminated.
 make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/facetimehd/0.1/build/fthd_ringbuf.o] Fehler 1
 make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/facetimehd/0.1/build/fthd_hw.o] Fehler 1
 make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/facetimehd/0.1/build/fthd_drv.o] Fehler 1
 make[1]: *** [/usr/src/linux-headers-6.8.0-38-generic/Makefile:1926: /var/lib/dkms/facetimehd/0.1/build] Fehler 2
 make: *** [Makefile:240: __sub-make] Fehler 2
 make: Verzeichnis „/usr/src/linux-headers-6.8.0-38-generic“ wird verlassen
DKMSKernelVersion: 6.8.0-38-generic
Date: Sun Jul 21 14:41:32 2024
Package: facetimehd-dkms 0.1
PackageVersion: 0.1
SourcePackage: facetimehd-dkms
Title: facetimehd-dkms 0.1: facetimehd kernel module failed to build

```

baumschubser, 2024-10-21:
Should be solved with commit 67bf8a73ed02c723d0177aaa7f85040f8991526f. Compiling works for me on kernel 6.8.0. Use the latest version and try again.

Hi-Angel, 2025-02-25:
@tr33 ping

rrthomas, 2026-09-12:
This issue can be closed; I've been happily using the driver with 6.8.0 on Ubuntu 24.04 for a couple of years!