patjak/facetimehd #299: dkms mkdeb doesn't work

vitacell, 2024-09-15:
```
vita@MacBookAir6-1:/usr/src/facetimehd-0.1$ sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
Error! Unknown action specified: ""
Usage: /usr/sbin/dkms [action] [options]
  [action]  = { add | remove | build | install | uninstall | match | autoinstall |
                mktarball | ldtarball | status }
  [options] = [-m module] [-v module-version] [-k kernel-version] [-a arch]
              [-c dkms.conf-location] [-q] [--force] [--force-version-override] [--all]
              [--templatekernel=kernel] [--directive='cli-directive=cli-value']
              [--config=kernel-.config-location] [--archive=tarball-location]
              [--kernelsourcedir=source-location]
              [--binaries-only] [--source-only] [--verbose]
              [--no-depmod] [--modprobe-on-install] [-j number] [--version]
vita@MacBookAir6-1:/usr/src/facetimehd-0.1$ sudo dkms mkdsc -m facetimehd -v 0.1 --source-only
Error! Unknown action specified: ""
Usage: /usr/sbin/dkms [action] [options]
  [action]  = { add | remove | build | install | uninstall | match | autoinstall |
                mktarball | ldtarball | status }
  [options] = [-m module] [-v module-version] [-k kernel-version] [-a arch]
              [-c dkms.conf-location] [-q] [--force] [--force-version-override] [--all]
              [--templatekernel=kernel] [--directive='cli-directive=cli-value']
              [--config=kernel-.config-location] [--archive=tarball-location]
              [--kernelsourcedir=source-location]
              [--binaries-only] [--source-only] [--verbose]
              [--no-depmod] [--modprobe-on-install] [-j number] [--version]
vita@MacBookAir6-1:/usr/src/facetimehd-0.1$ sudo dkms mkdeb -m facetimehd -v 0.1 --source-only
Error! Unknown action specified: ""
Usage: /usr/sbin/dkms [action] [options]
  [action]  = { add | remove | build | install | uninstall | match | autoinstall |
                mktarball | ldtarball | status }
  [options] = [-m module] [-v module-version] [-k kernel-version] [-a arch]
              [-c dkms.conf-location] [-q] [--force] [--force-version-override] [--all]
              [--templatekernel=kernel] [--directive='cli-directive=cli-value']
              [--config=kernel-.config-location] [--archive=tarball-location]
              [--kernelsourcedir=source-location]
              [--binaries-only] [--source-only] [--verbose]
              [--no-depmod] [--modprobe-on-install] [-j number] [--version]
vita@MacBookAir6-1:/usr/src/facetimehd-0.1$ sudo dkms mkdeb -m facetimehd -v 0.1 --source-only
```

giordy, 2024-10-26:
yes I stumbled upon this too...
Unfortunately dkms has removed support for distro-specific actions in the latest releases.
I'd be also interested to know how to fix this.

vitacell, 2024-10-29:
> yes I stumbled upon this too... Unfortunately dkms has removed support for distro-specific actions in the latest releases. I'd be also interested to know how to fix this.

Angry rtrded devs like breaking GNU/Linux experience. I can not to explain why they like to break userspace. They break and remove everything, and then give no solution to users. So, now, what we supposed to do or use instead?

giordy, 2024-10-29:
@vitacell in my case I solved by using this PPA https://launchpad.net/~greg-whiteley/+archive/ubuntu/facetimehd

vitacell, 2024-10-29:
> @vitacell in my case I solved by using this PPA https://launchpad.net/~greg-whiteley/+archive/ubuntu/facetimehd

Hey, thanks, will try. Because after every Ubuntu update/bugdate, I have to reinstall Broadblob Facetimehd crap manually.

NostalgiaRealm, 2025-03-05:
> [@vitacell](https://github.com/vitacell) in my case I solved by using this PPA https://launchpad.net/~greg-whiteley/+archive/ubuntu/facetimehd

This just saved me spending hours on troubleshooting this issue. It wasn't mentioned in any of the searches I did online. Thank you big time!

vrilutza, 2026-10-17:
`mkdeb` and `mkdsc` were removed in DKMS 3.0.0 (dkms-project/dkms@68b083eaa3f7), which is why dkms reports them as an unknown action. A .deb isn't needed for automatic rebuilds: after `dkms add`/`build`/`install` once, DKMS rebuilds the module when a new kernel is installed together with its headers (`linux-headers-generic` on Ubuntu brings them in), so it doesn't have to be reinstalled by hand after an update.

The Installation page now describes it this way: https://github.com/patjak/facetimehd/wiki/Installation#setting-up-dkms-auto-compile-on-kernel-update

If this answers the question, feel free to close the issue.