patjak/facetimehd #301: Driver doesn't work on latest Mint/Ubuntu

eugenialoli, 2024-11-23:
After successfully installing the firmware and the driver I get this

```
eugenia@MacBookAir-2015:/usr/lib/firmware/facetimehd$ v4l2-ctl --list-devices
Apple Facetime HD (PCI:0000:02:00.0):
	/dev/video0

```
However, while the driver is loaded, it doesn't work:

```
eugenia@MacBookAir-2015:/usr/lib/firmware/facetimehd$ cheese 

(cheese:3556): cheese-WARNING **: 18:22:54.026: stream error: can't negotiate buffers on port: ../src/gst/gstpipewiresrc.c(692): on_state_changed (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin36/GstPipeWireSrc:pipewiresrc1

```

And OBS-Studio:

```
info: v4l2-input: Start capture from 
error: v4l2-input: Unable to open device
error: v4l2-input: Initialization failed, errno: No such file or directory
info: v4l2-input: Start capture from 
error: v4l2-input: Unable to open device
error: v4l2-input: Initialization failed, errno: No such file or directory
info: v4l2-input: Start capture from 
error: v4l2-input: Unable to open device
error: v4l2-input: Initialization failed, errno: No such file or directory
info: Switched to scene 'Scene'
info: ------------------------------------------------
info: Loaded scenes:
info: - scene 'Scene':
info:     - source: 'Video Capture Device (V4L2)' (v4l2_input)
info:     - source: 'Video Capture Device (V4L2) 2' (v4l2_input)
info:     - source: 'Video Capture Device (V4L2) 3' (v4l2_input)
info: ------------------------------------------------
info: adding 21 milliseconds of audio buffering, total audio buffering is now 21 milliseconds (source: Desktop Audio)

info: v4l2-input: Start capture from 
error: v4l2-input: Unable to open device
error: v4l2-input: Initialization failed, errno: No such file or directory
info: User added source 'Video Capture Device (V4L2) 4' (v4l2_input) to scene 'Scene'
free(): invalid pointer
```

rodstar97, 2025-01-26:
Have the exact same Problem.
System:
  Kernel: 6.8.0-51-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.6 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: Apple product: MacBookAir6,1 v: 1.0 serial: <superuser required> Chassis:
    type: 10 v: Mac-35C1E88140C3E6CF serial: <superuser required>
  Mobo: Apple model: Mac-35C1E88140C3E6CF v: MacBookAir6,1 serial: <superuser required>
    uuid: <superuser required> UEFI: Apple v: 474.0.0.0.0 date: 08/23/2022
Battery:
  ID-1: BAT0 charge: 11.4 Wh (46.9%) condition: 24.3/38.8 Wh (62.7%) volts: 8.3 min: 7.6
    model: DP bq20z451 type: Li-ion serial: N/A status: charging cycles: 917
CPU:
  Info: dual core model: Intel Core i5-4260U bits: 64 type: MT MCP smt: enabled arch: Haswell
    rev: 1 cache: L1: 128 KiB L2: 512 KiB L3: 3 MiB
  Speed (MHz): avg: 1600 min/max: 800/2700 cores: 1: 1600 2: 1600 3: 1600 4: 1600 bogomips: 16000
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel Haswell-ULT Integrated Graphics vendor: Apple driver: i915 v: kernel
    arch: Gen-7.5 ports: active: eDP-1 empty: DP-1, DP-2, HDMI-A-1, HDMI-A-2 bus-ID: 00:02.0
    chip-ID: 8086:0a26 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: modesetting
    unloaded: fbdev,vesa dri: crocus gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1366x768 s-dpi: 96 s-size: 361x203mm (14.21x7.99") s-diag: 414mm (16.31")
  Monitor-1: eDP-1 model: Apple Color LCD res: 1366x768 hz: 60 dpi: 133
    size: 260x140mm (10.24x5.51") diag: 295mm (11.6") modes: 1366x768
  API: EGL v: 1.5 hw: drv: intel crocus platforms: device: 0 drv: crocus device: 1 drv: swrast
    surfaceless: drv: crocus x11: drv: crocus inactive: gbm,wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 24.0.9-0ubuntu0.3 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel HD Graphics 5000 (HSW GT3) device-ID: 8086:0a26
Audio:
  Device-1: Intel Haswell-ULT HD Audio vendor: Apple driver: snd_hda_intel v: kernel
    bus-ID: 00:03.0 chip-ID: 8086:0a0c class-ID: 0403
  Device-2: Intel 8 Series HD Audio driver: snd_hda_intel v: kernel bus-ID: 00:1b.0
    chip-ID: 8086:9c20 class-ID: 0403
  Device-3: Broadcom 720p FaceTime HD Camera driver: facetimehd v: kernel pcie: speed: 5 GT/s
    lanes: 1 bus-ID: 02:00.0 chip-ID: 14e4:1570 class-ID: 0480
  API: ALSA v: k6.8.0-51-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Broadcom BCM4360 802.11ac Dual Band Wireless Network Adapter vendor: Apple driver: wl
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 bus-ID: 03:00.0 chip-ID: 14e4:43a0 class-ID: 0280
  IF: wlp3s0 state: up mac: <filter>
Bluetooth:
  Device-1: Apple Bluetooth USB Host Controller driver: btusb v: 0.8 type: USB rev: 2.0
    speed: 12 Mb/s lanes: 1 bus-ID: 1-3.3:6 chip-ID: 05ac:828f class-ID: fe01
  Report: hciconfig ID: hci0 rfk-id: 2 state: up address: <filter> bt-v: 4.0 lmp-v: 6 sub-v: 4196
    hci-v: 6 rev: 2466 class-ID: 7c010c
Drives:
  Local Storage: total: 113 GiB used: 14.49 GiB (12.8%)
  ID-1: /dev/sda vendor: Apple model: SSD TS0128F size: 113 GiB speed: 6.0 Gb/s tech: SSD
    serial: <filter> fw-rev: 0219 scheme: GPT
Partition:
  ID-1: / size: 57.03 GiB used: 14.48 GiB (25.4%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 3.87 GiB used: 256 KiB (0.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: hi-speed hub with single TT ports: 9 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-2: 1-3:2 info: Broadcom BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth) ports: 3 rev: 2.0
    speed: 12 Mb/s lanes: 1 power: 94mA chip-ID: 0a5c:4500 class-ID: 0900
  Device-1: 1-3.3:6 info: Apple Bluetooth USB Host Controller type: bluetooth driver: btusb
    interfaces: 4 rev: 2.0 speed: 12 Mb/s lanes: 1 chip-ID: 05ac:828f class-ID: fe01
  Device-2: 1-5:3 info: Apple Apple Internal Keyboard / Trackpad type: HID,keyboard,mouse
    driver: apple,bcm5974,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s lanes: 1 power: 500mA
    chip-ID: 05ac:0290 class-ID: 0301 serial: <filter>
  Hub-3: 2-0:1 info: super-speed hub ports: 4 rev: 3.0 speed: 5 Gb/s lanes: 1 chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: 64.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Repos:
  Packages: pm: dpkg pkgs: 2017
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/greg-whiteley-facetimehd-noble.list
    1: deb [signed-by=/etc/apt/keyrings/greg-whiteley-facetimehd-noble.gpg] https: //ppa.launchpadcontent.net/greg-whiteley/facetimehd/ubuntu noble main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb https: //ftp.fau.de/mint/packages xia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu noble main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu noble-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu noble-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ noble-security main restricted universe multiverse
Info:
  Memory: total: 4 GiB available: 3.77 GiB used: 2.5 GiB (66.5%)
  Processes: 223 Power: uptime: 30m states: freeze,mem,disk suspend: deep wakeups: 0
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Cinnamon v: 6.4.6 inxi: 3.3.34

da0ist, 2025-02-01:
[~] jeward@mbp12-1 $ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=25.04
DISTRIB_CODENAME=plucky
DISTRIB_DESCRIPTION="Ubuntu Plucky Puffin (development branch)"
PRETTY_NAME="Rhino Linux 2025.1 (mainline)"
NAME="Rhino Linux"
VERSION_ID="2025.1"
VERSION="2025.1 (mainline)"
VERSION_CODENAME="devel"
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://rhinolinux.org/"
SUPPORT_URL="https://github.com/rhino-linux"
BUG_REPORT_URL="https://github.com/rhino-linux"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME="devel"

[~] jeward@mbp12-1 $ cheese

(cheese:4759): Gdk-CRITICAL **: 05:23:00.880: gdk_wayland_window_set_dbus_properties_libgtk_only: assertion 'GDK_IS_WAYLAND_WINDOW (window)' failed

(cheese:4759): cheese-WARNING **: 05:23:01.237: stream error: can't negotiate buffers on port: ../src/gst/gstpipewiresrc.c(704): on_state_changed (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin28/GstPipeWireSrc:pipewiresrc1

Hi-Angel, 2025-02-25:
This is most likely due to some in-kernel API change.

I see nobody's digging into this, but at the same time I see pull-requests are being merged, so presumably if you find which kernel change broke this driver, it may hopefully be easily fixed.

So, if somebody has motivation, you'd need to bisect the kernel. First of all, try checking with, like, 6.6 or 6.1 to make sure older versions indeed worked. Then you'd need to build the kernel yourself for commits between the "bad" and "good" range *(marked by `git bisect bad` and `git bisect good`)*, and don't forget to recompile the driver. There's also `git bisect skip` in case some commit doesn't work.

Bisecting isn't that long, it is O(log(n)) in terms of complexity, so e.g. for 1000 commits you can find the culprit in 10 steps.

pschatzmann, 2026-10-01:
I have a [working fork](https://github.com/pschatzmann/facetimehd). 
The [pull request](https://github.com/patjak/facetimehd/pull/355) has been submitted...

vrilutza, 2026-10-16:
#355 from pschatzmann is merged now, so current master has that fix.