patjak/facetimehd #303: Compiling facetimehd on CachyOS

krzyczak, 2025-01-27:
I tried compiling this on CachyOS (which is based on Arch) https://cachyos.org and I run into some issues, however I managed to get it working.

Here is what I did:
1. CachyOS kernel is compiled with clang instead of gcc, so first thing to note is that `make` needed to be run like this:
```bash
make CC=clang CXX=clang++ LD=ld.lld
```
2.

nicahokie, 2025-01-27:
As you say I think point 1 might just be specific to CachyOS

but I think your point 2 is necessary for other distros on kernel 6.13+, but was not for kernel 6.12.x
In my case I am on Xanmod kernel 6.13 on ubuntu 24.10 and it helped me

vrilutza, 2026-10-16:
Fixed in master by #304 (the prandom.h include for 6.13+).