patjak/facetimehd #306: build failure on arch current kernel 6.13.4

jbholland, 2025-02-28:
Hi, your package is great, it is not building on my just-updated Arch install though. Attached is the make.log file that seems to indicate some problems. It is working on "lts" long-term-support which is 6.12.16-1. 

[make.log](https://github.com/user-attachments/files/19028857/make.log)

y-tzk, 2025-03-01:
have you tried this patch?

https://github.com/patjak/facetimehd/compare/master...jakintosh:facetimehd:master

thatsthat, 2025-03-03:
> have you tried this patch?
> 
> [master...jakintosh:facetimehd:master](https://github.com/patjak/facetimehd/compare/master...jakintosh:facetimehd:master)

I can confirm the patch fixed this issue in my system.

vrilutza, 2026-10-16:
Fixed in master by #304 (the prandom.h include for 6.13+).