patjak/facetimehd #310: "No such file or directory" during checkinstall (dkms subfolder in modules/.../updates ??)

rauferd, 2025-05-14:
facetimehd 0.6.13 on Debian 13, linux kernel 6.12.27.
After successful compilation, I'd like to create a .deb package with checkinstall. Expected behaviour: No error message, funcational .deb file created.

Instead, checkinstall throws an error, see below for details:
`xz: /lib/modules/6.12.27-amd64/updates/facetimehd.ko: No such file or directory`

The file does not exist, but there is `/lib/modules/6.12.27-amd64/updates/dkms/facetimehd.ko` (note the extra subfolder `dkms`). What's going wrong? How can I tell checkinstall to find the module?

Previously, I did not see this error, but I can't tell whether it's related to the Debain update 12->13 or the facetimehd update - so I don't even know if it's apropriate to report the issue here?

Full console output:
```
root:/usr/src/facetimehd-0.6.13# checkinstall 

checkinstall 1.6.3, Copyright 2010 Felipe Eduardo Sanchez Diaz Duran
           This software is released under the GNU GPL.

*****************************************
**** Debian package creation selected ***
*****************************************

This package will be built according to these values: 

0 -  Maintainer: [ root ]
1 -  Summary: [ FacetimeHD Driver ]
2 -  Name:    [ facetimehd ]
3 -  Version: [ 0.6.13 ]
4 -  Release: [ 1 ]
5 -  License: [ GPL ]
6 -  Group:   [ checkinstall ]
7 -  Architecture: [ amd64 ]
8 -  Source location: [ facetimehd-0.6.13 ]
9 -  Alternate source location: [  ]
10 - Requires: [  ]
11 - Recommends: [  ]
12 - Suggests: [  ]
13 - Provides: [ facetimehd ]
14 - Conflicts: [  ]
15 - Replaces: [  ]
16 - Prerequires: [  ]

Enter a number to change any of them or press ENTER to continue: 

Installing with make install...

========================= Installation results ===========================
make -C /lib/modules/6.12.27-amd64/build M=/usr/src/facetimehd-0.6.13 modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.12.27-amd64'
  INSTALL /lib/modules/6.12.27-amd64/updates/facetimehd.ko
  SIGN    /lib/modules/6.12.27-amd64/updates/facetimehd.ko
  XZ      /lib/modules/6.12.27-amd64/updates/facetimehd.ko.xz
xz: /lib/modules/6.12.27-amd64/updates/facetimehd.ko: No such file or directory
make[3]: *** [/usr/src/linux-headers-6.12.27-common/scripts/Makefile.modinst:159: /lib/modules/6.12.27-amd64/updates/facetimehd.ko.xz] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.27-common/Makefile:1882: modules_install] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.27-common/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.27-amd64'
make: *** [Makefile:18: install] Error 2

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.
```

lymber, 2025-08-11:
Hi, everyone!

Just to confirm the problem mentioned above. Now with the Debian 13 release, I had to recompile everything and the checkinstall showed the following:

checkinstall 1.6.3, Copyright 2010 Felipe Eduardo Sanchez Diaz Duran
           This software is released under the GNU GPL.

*****************************************
**** Debian package creation selected ***
*****************************************

This package will be built according to these values: 

0 -  Maintainer: [ root@macbookpro ]
1 -  Summary: [ FaceTime HD driver ]
2 -  Name:    [ facetimehd ]
3 -  Version: [ 20250811 ]
4 -  Release: [ 1 ]
5 -  License: [ GPL ]
6 -  Group:   [ checkinstall ]
7 -  Architecture: [ amd64 ]
8 -  Source location: [ facetimehd ]
9 -  Alternate source location: [  ]
10 - Requires: [  ]
11 - Recommends: [  ]
12 - Suggests: [  ]
13 - Provides: [ facetimehd ]
14 - Conflicts: [  ]
15 - Replaces: [  ]
16 - Prerequires: [  ]

Enter a number to change any of them or press ENTER to continue: 

Installing with make install...

========================= Installation results ===========================
make -C /lib/modules/6.12.38+deb13-amd64/build M=/home/lymber/tmp/facetimehd modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.12.38+deb13-amd64'
  INSTALL /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko
  SIGN    /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko
At main.c:171:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: ../crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: ../crypto/bio/bss_file.c:75
sign-file: /usr/src/linux-headers-6.12.38+deb13-common/certs/signing_key.pem
  XZ      /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko.xz
xz: /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko: No such file or directory
make[3]: *** [/usr/src/linux-headers-6.12.38+deb13-common/scripts/Makefile.modinst:159: /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko.xz] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.38+deb13-common/Makefile:1878: modules_install] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.38+deb13-common/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.38+deb13-amd64'
make: *** [Makefile:18: install] Error 2

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.

The, with with the "make install" procedure, I got:

make -C /lib/modules/6.12.38+deb13-amd64/build M=/home/lymber/tmp/facetimehd modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.12.38+deb13-amd64'
  INSTALL /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko
  SIGN    /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko
At main.c:171:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: ../crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: ../crypto/bio/bss_file.c:75
sign-file: /usr/src/linux-headers-6.12.38+deb13-common/certs/signing_key.pem
  XZ      /lib/modules/6.12.38+deb13-amd64/updates/facetimehd.ko.xz
  DEPMOD  /lib/modules/6.12.38+deb13-amd64
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.12.38+deb13-amd64'

After, this (with the same SSL errors, I run 'depmod' and 'modprobe facetimehd' and the camera was on.

I am available to run some tests and provide more information if needed.

Best, Alexandre

akamiya, 2025-10-16:
Have the same result as above on MacBookPro Mid-2014

I am available to run some tests and provide more information if needed, as well.