patjak/facetimehd #314: [Request] FaceTime HD Camera Firmware for MacBook Pro 14,1 (Mid-2017)

ashnonokplays, 2025-06-22:
Hi everyone,

I'm currently running Linux Mint 22.1 on a MacBook Pro 14,1 (13-inch, Mid-2017) and have successfully installed the mbp-2016-linux patched kernel to get audio and keyboard/trackpad working properly.

However, I'm stuck on getting the FaceTime HD webcam working. I’ve already built and installed the bcwc_pcie driver, but as many of you know, the driver requires a firmware.bin file extracted from macOS (AppleCameraInterface.kext), which I no longer have access to.

    My Mac's SSD has been fully wiped

    Internet Recovery is failing, likely due to slow connection or server deprecation

    I don’t have another Mac to borrow the kext from

If someone has already extracted the firmware from AppleCameraInterface.kext (from macOS High Sierra or newer) and is using it with a similar model (MacBook Pro 14,1), I’d really appreciate it if you could share the firmware.bin file with me privately or guide me where to find it legally.

I understand that sharing Apple firmware is a legal gray area, so if this post violates any rules, I’ll be happy to take it down or follow private instructions.

Thanks so much in advance for any help!

—

System Info:

    MacBookPro14,1 (13-inch, 2017, no T2 chip)

    Linux Mint 22.1 (based on Ubuntu 22.04)

    Kernel: 6.x-mbp+ (via mbp-2016-linux precompiled .deb)

    Driver installed: bcwc_pcie (built from GitHub)

BrewCoffeeAddict, 2025-07-06:
Hello @ashnonokplays 

I also have a MacBook Pro 14,1 but I am on Ventura and I can't get the firmware file from Mac OS X. I believe they have removed AppleCameraInterface.kext in newer version of Mac OS X.
Instead, you can get the firmware.bin from Bootcamp by following these instructions: https://github.com/patjak/facetimehd/wiki/Get-Started#firmware-extraction

I hope this helps

ashnonokplays, 2025-07-31:
as I've told before I no longer have access to Mac OS I've wipe it clean and my internet connection is not that strong.